wled/WLED #3815: Widgets for Android ?

sci666-2023, 2024-03-12:
can you add Widget Support for Android with on/off buttons and maybe macro/preset buttons ? that would be great to make a simple Tablet or Smartphone  "Control Screen"

blazoncek, 2024-03-12:
Please open an issue in the WLED app repository. WLED doesn't do Android or iOS.

BTW WLED app has been discontinued. A new app has been developed ba @Moustachauve .

closed 2024-03-12 by blazoncek

Aircoookie, 2024-03-12:
The app [HTTP Request Shortcuts](https://http-shortcuts.rmy.ch/) is also amazing for this use case. I use shortcuts on my home screen for 90% of WLED controlling I do from my phone. Of course you need to look at the WLED API for the relevant commands (e.g. `ip/win&T=2` for toggle on/off)

sci666-2023, 2024-03-12:
> Please open an issue in the WLED app repository. WLED doesn't do Android or iOS.
> 
> BTW WLED app has been discontinued. A new app has been developed ba @Moustachauve .

Which App is it and where can I find it?

blazoncek, 2024-03-12:
WLED Native