wled/WLED #3817: Static IP not saving and DHCP been used instead

Doyle4, 2024-03-12:
### What happened?

Chose my static IP, press saved, DHCP created one

### To Reproduce Bug

Enter static IP and press save

### Expected Behavior

To create a static IP 
[image]

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2 - 8266 - 160mhz

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-12:
Please provide a debug output (boot sequence) using debug builds from @srg74 .

Doyle4, 2024-03-14:
Not got back yet sorry as I have a feeling its my router causing the issue.
14.1/b1 was working fine setting a static IP and now devices on older firmwares are doing the same, so I don't think its WLED and more the case of my router not accepting a static IP I set as its conflicting.

Going to remove all devices from router and see if that solves it, if not, will get the debug build and test.

farmmanic, 2024-03-17:
you can unplug the router to reset the dhcp leases. i recommend using ip addresses in the higher range of the dhcp net. maybe starting at 200. as routers will enumerate addresses starting at the beginning of the nets range.

Doyle4, 2024-03-18:
> you can unplug the router to reset the dhcp leases. i recommend using ip addresses in the higher range of the dhcp net. maybe starting at 200. as routers will enumerate addresses starting at the beginning of the nets range.

Will try in a higher range thanks

closed 2024-07-25 by github-actions[bot]