wled/WLED #3825: WLED output #3 not working, and #4 not configuring on ESP32-C3

Spire9, 2024-03-15:
### What happened?

WLED will not correctly configure all channels on the ESP32-C3.  Mixing up the channel order confirms the hardware tests all good.   WLED channels #3 and #4 can not control LED strips.

### To Reproduce Bug

Try to configure four channels at once.  1&2 work great.  #3 pretends to configure, but fails to send any data.  #4 does not even offer a proper configuration in the dropdown.  See screenshot: https://i.ibb.co/HPDz7JZ/wled-settings.png 

### Expected Behavior

All 4 channels should be available.  I know the C3's clock speed of 160mhz is a bit lower than the WROOM's 240mhz, but not so much it should be locked out for those of us that don't even run effects.  It must be a bug.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.2-b1 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Relevant log/trace output

```shell
https://i.ibb.co/v18Wfq7/wled-info.png
https://i.ibb.co/HPDz7JZ/wled-settings.png
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-15:
Channels? Perhaps you're talking about buses (outputs).
ESP32-C3 only has 2 RMT buffers and can only drive 2 digital LED outputs. It lacks support for I2S for LEDs. It also has 6 LEDC channels so can drive 1 RGBW(+CCT) or 2 RGB analog outputs.

WLED cannot overcome hardware limitations.

closed 2024-03-15 by blazoncek

Spire9, 2024-03-15:
Right, that.  So now the conversation is at the point where we both understand the screenshot shows an off-by-one error at output #3 in WLED.