wled/WLED #3826: Wled mqtt status updating slow

imawizzard, 2024-03-15:
### What happened?

I send commands to my wled over mqtt. I check on mqtt if the wled is online, Why is it taking so long for wled to send the status online? If I send an effect. My wled changes to the effect, but says it is still offline. It is reacting, so it is online right?

### To Reproduce Bug

Send a command over mqtt

### Expected Behavior

When I send a command to my wled, end it received it, it should go online...

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2-b2

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-15:
WLED does not send "online" messages. When it connects to MQTT broker it sends "online" message and sets LWT as "offline".
There are no subsequent "online" messages unless reconnect to broker happens.

imawizzard, 2024-03-15:
[image]

blazoncek, 2024-03-16:
@imawizzard "status" is a retained payload with a LWT as explained above.
It is only ever retransmitted when a WLED state changes. This may be within seconds or never.

softhack007, 2024-03-16:
@blazoncek @Aircoookie -  not strictly related to this topic - I just saw that our copy of async-mqtt-client is 5 years old, while the [original lib](https://github.com/marvinroger/async-mqtt-client/releases) has received some updates in the meantime (release 0.9.0 in July 2021) . 

Would it make sense to remove the old copy from wled00/src/dependencies/async-mqtt-client, and use a real dependency (lib_deps) to the original repo instead? Or we have a good reason to stay with our older copy?

imawizzard, 2024-03-16:
But, when I send my effect and wled receive it. Why doesn't send "online". Is there an other way to check if wled is online?

blazoncek, 2024-03-16:
It does on my system but in any case it is a retained message, there are no needs for it to be retransmitted as it has LWT attached which will be returned when a device is no longer connected to a broker.
[image]

@softhack007 there were no changes to the library in last 3 years. Did you find a discrepancy?

imawizzard, 2024-03-16:
> It does on my system but in any case it is a retained message, there are no needs for it to be retransmitted as it has LWT attached which will be returned when a device is no longer connected to a broker. ![Screenshot 2024-03-16 at 12 40 33](https://private-user-images.githubusercontent.com/7722392/313411845-35472847-072d-4f72-b762-8c79ec9bab63.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2MDA5OTUsIm5iZiI6MTcxMDYwMDY5NSwicGF0aCI6Ii83NzIyMzkyLzMxMzQxMTg0NS0zNTQ3Mjg0Ny0wNzJkLTRmNzItYjc2Mi04Yzc5ZWM5YmFiNjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTZUMTQ1MTM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGQ5YTYwNDYzOTg4NDE1ZTQxNDI5MDNmYTQwNTg1NzY1MjM2NWMxZmE3OTY5ZjAyNDY5NzFiNDk4M2VmZWJhYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.CXvZ3Z4oz7aqyy1wgfBSE9hsgEeaxtp2POGb4BV-goU)
> 
> @softhack007 there were no changes to the library in last 3 years. Did you find a discrepancy?

But when I restart my device, it takes 2 minuten to go back online

softhack007, 2024-03-16:
> @softhack007 there were no changes to the library in last 3 years. Did you find a discrepancy?

Did not diff the source code yet. But it clear our copy was last updated on Oct 20, 2019; while the original one has some changes between Feb 2021 and July 2021 that we don't have.

https://github.com/marvinroger/async-mqtt-client/commits/develop/

I'm not an expert on MQTT, so can't say if the changes are relevant for WLED use cases.

softhack007, 2024-03-16:
> But when I restart my device, it takes 2 minuten to go back online

@imawizzard are you sending a retained message (aka Willi)? 
😉 sorry could not resist. I have the feeling you already made this statement.

blazoncek, 2024-03-16:
> But when I restart my device, it takes 2 minuten to go back online

Did you read my response? It is a retained message. Retained messages are transmitted from broker when a client subscribes to a topic.