wled/WLED #3829: Please provide binaries for the 4MB version of the ESP32-S3

mase76, 2024-03-17:
Hi,
could you please provide binaries for the 4MB version of the EPS32-S3? Me and a friend were able to built them, and we can confirm, that is works.

blazoncek, 2024-03-17:
I do not own one to test. Perhaps @softhack007 has it?

mase76, 2024-03-17:
I have designed a PCB with this device. Maybe I can send you one.

blazoncek, 2024-03-17:
Thanks. PM me on Discord.
BTW you can use `board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv` in the build environment.

softhack007, 2024-03-18:
> I do not own one to test. Perhaps @softhack007 has it?

Yes, I have one -S3 with 4MB flash and 4M  PSRAM for testing. I'm currently on holidays and "away from wled", so I'd need to look into it in beginning of April.

softhack007, 2024-03-18:
Edit: I think the inofficial installer installer from @wled-install  (Wladi) already has a build for -S3 4MB. You might find your firmware there until we add it to out official builds.

https://wled-install.github.io/

@blazoncek we also need to add "official release" build for -S3 with 4MB PSRAM (qspi) and >= 8MB PSRAM (opi) - as these variants both need a separate firmware build and will not work with other -S3 builds.

mase76, 2024-03-18:
I use a ESP32-S3FH4R2 with 2MB PSRAM. I think, there is also a different bootloader needed.

blazoncek, 2024-04-10:
ba9ce4adf2adc0261bd2572e220f413860313a69

closed 2024-06-23 by blazoncek

SunboX, 2024-11-07:
@mase76 I also have a ESP32-S3FH4R2 with 2MB PSRAM. Can you please share more details how you did build WLED for it?

softhack007, 2024-11-07:
@mase76 the 4MB S3 board should work with this standard firmware image from the release page:
* https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b7/WLED_0.15.0-b7_ESP32-S3_4M_qspi.bin