wled/WLED #3832: strip doesn't turn after update

R3Cardoso, 2024-03-18:
### What happened?

If I update my ESP32 Gledopto controller from anything after 1.4.0, the strip doesn't work after power is off.
If I use 1.4.0 there is not problem, if I do a factory reset on anything after 1.4.0 it work, but as soon I power off the power it doesn't turn on the strip any more.

### To Reproduce Bug

Strip doesn't light up after losing power on any update after 1.4.0.

### Expected Behavior

turn on strip, as it does on version 1.4.0

### Install Method

Binary from WLED.me

### What version of WLED?

1.4.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-18:
Please provide serial output of boot sequence.
You may also want to upload debug binary.
You may also want to erase flash entirely using esptool or similar.

R3Cardoso, 2024-03-18:
I don't know how to get that.
But if I put it on version 1.4.0, everything works fine.
If I do a factory reset it works fine until I power it off(remote power source)

blazoncek, 2024-03-18:
Do a flash erase and then upload newer binary.

R3Cardoso, 2024-03-18:
the board doesn't have a usb power, and I don't have tools to connect direclty to board.
it's possible to do it wireless?

blazoncek, 2024-03-18:
no. unfortunately that means I'm unable to help further.

R3Cardoso, 2024-03-18:
I undestand it.
It's strange how it works fine on 1.4.0, it works fine after a factory reset on any other upgrade, but only as soon it lost power it doesn't work any more.

blazoncek, 2024-03-19:
yes, that is fixed in 0.15

closed 2024-03-29 by blazoncek