wled/WLED #3838: 0.14.2 binaries are actually 0.14.2-b2 (or their names are wrong)

B1BU, 2024-03-20:
### What happened?

The binaries in the 0.14.2 release are named 0.14.2-b2. I don't know if this was a typo or if you guy's accidentally uploaded the wrong version

### To Reproduce Bug

Check [Releases](https://github.com/Aircoookie/WLED/releases)
0.14.2 and 0.14.2-b2 seem to have some of the same files

### Expected Behavior

For the files to be listed as 0.14.2 without the b2 suffix I guess

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-20:
There are no differences in code between 0.14.2-b2 and 0.14.2.
However I did make a mistake by not pushing local changes before making a release. Uploaded files are from local build that was 0.14.2.

blazoncek, 2024-03-20:
I fixed double upload. Thank you for pointing it out.

closed 2024-03-20 by blazoncek