wled/WLED #3841: JSON | palettes key null or missing if more than 6 segments

smirgol, 2024-03-21:
### What happened?

I have 9 display with 32 leds each, controlled by a Wemos D1 mini.

After upgrading from a somewhat old firmware - I think some 1.2 or 1.3 - to either the latest version (1.5) or the latest 1.4 version, I experience an issue with fetching the JSON data.

When I add the segments, 9 in total with each one containing 32 LEDs, when adding Segment 6 the JSON data will start to contain a `null` value for the key `palettes`. When adding even more than 8 segments, the key `palettes` will completely disappear in the JSON data.

### To Reproduce Bug

* Fresh install of a recent version of WLED on a D1 mini
* add more than 6 segments
* check result of `http://xx.xx.xx.xx/json` endpoint - `palettes` is either `null` or missing

### Expected Behavior

`palettes` should be in the JSON data, independently of how many segments are configured.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.2 (build 2403170)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-21:
ESP8266 has limited amount of RAM and many segments may overflow JSON buffer when fetching /json.
You can compile your own version and change JSON_BUFFER_SIZE in const.h to increase buffer size. Unfortunately that may have adverse effect on stability.

I would recommend to upgrade to ESP32 instead. Or use /json/palettes on its own.

smirgol, 2024-03-21:
I see, I assumed something like that, but thought maybe it's a regression.
I think I'll try to downgrade then, as it did work in the older version. Don't need the fancies the newer versions bring. :)
Thanks for your feedback!

blazoncek, 2024-03-21:
You may want to try 0.15.0-b1 as it improved on available RAM.
JSON buffer is still limited to 10kB but by custom compilation it could be safely increased to 12kB.

closed 2024-07-27 by github-actions[bot]