wled/WLED #3845: ESP8266 NodeMCU v3 connection lost after turning on LEDs

Xitee1, 2024-03-22:
### What happened?

As soon as the light is turned on in v0.14.1-b1 and above on a NodeMCU v3 board, a connection lost message is displayed and the web page is very very laggy. If you're lucky you are able to turn off the light again before the connection drops completely and WLED stays reachable as normal.
In v0.14.0 everything is working. When the light is on the device stays reachable as normal.

Now the very weird thing is that I had no issues with v0.14.1 (or maybe the v0.14.2 betas - don't remember exactly) before. Only after updating to v0.14.2 broke the things. Downgrading to v0.14.1 weirdly did not help at all (even with restoring config from backup). Now I have to stay on v0.14.0 to keep my LEDs usable.
Other boards like the wemos d1 mini or s2 mini are not affected and work correctly in the newest version.

### To Reproduce Bug

See above.

### Expected Behavior

WLED staying reachable even though the LEDs are on.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Using the new 160 build slightly helped keeping the device reachable. But it is still very very slow in response and after around a minute the connection is also completely lost. I think something extremely overloads the ESP.
The latest beta [v0.15.0-b1](https://github.com/Aircoookie/WLED/releases/tag/v0.15.0-b1) also has this problem.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

craigwitter, 2024-04-01:
I'm seeing the same. Going back to 14.0 fixes it for me, also only seeing the issue on the 8266.

blazoncek, 2024-04-01:
This is usually an issue of on-board PSU not being able to supply enough current to ESP chip.
Use a dedicated 3.3V PSU for ESP.

closed 2024-04-01 by blazoncek

Xitee1, 2024-04-01:
My ESP is connected to a dedicated 5V USB power source. And it doesn't make sense that a software update causes this issue then. If it was a hardware issue, every software version should have this issue. However it was introduced in 14.1. Earlier versions do work just fine.
Also, my two NodeMCU v3 boards are used in different setups with different wiring and LEDs types.

blazoncek, 2024-04-01:
> If it was a hardware issue, every software version should have this issue.

Not necessarily. New features require more CPU power and hence greater current demand.
5V needs to be converted to 3.3V for ESP to use, that is done by on-board PSU (5V->3.3V). Low quality devices have inadequate current capabilities. You can verify this by monitoring serial output and check foe brownouts or blackuouts.

If you insist on being a SW issue then justify how it works on one device and not on the other? Binary image is the same.
If you still insist on being a SW issue then please provide crash dumps using debug builds and exception decoder.