wled/WLED #3846: ESP32 showing "Brownout detector was triggered" - possible defect?

Panossa, 2024-03-22:
### What happened?

I installed WLED V 0.14.2 on my ESP and it worked fine. I could set up everything, even tested out the touch features etc. Then, suddenly, it would not start a webservice any more. 

Around that time I wanted to try out new LED strips and wired the power and ground in the wrong config (GND pin to power of the strip and 5V pin to ground of the strip). 

Before that, I triggered a non-existing preset via touch commands (I told it to use the preset with index 3 but there was none with that index) so I think it should not even have sent voltage down the pins since there was no config set. Nonetheless, I smelled burning plastic after a few seconds and turned it off. Removed the connected strip and tried to start it up. 

Since then I couldn't get it to start up the web service and I noticed a few things:

1. Overriding the OS on the ESP via install.wled.me works every time except the first I tried to today (had to press the boot button to reset it "hard"). It erases all data, writes the WLED software to the ESP and says it finished with success.
2. All Wi-Fi functionality doesn't seem to work. Neither does the ESP open a Wi-Fi AP, nor does "connect to Wi-Fi" get any results when trying to fetch available networks.
3. The logs I can see via install.wled.me show the following error, repeating every 1-2 seconds:

[image]

### To Reproduce Bug

No consistency...

### Expected Behavior

WLED starting up in any shape or form.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
```

### Anything else?

The strip I tried to use is a 1m 144led/m strip from BTF lighting.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Panossa, 2024-03-22:
One small addition: I've seen a few posts talking about this error. A few mentioned a possible hardware defect, most had the same error (with SW_CPU_RESET and SPI_FAST_FLASH_BOOT) but with "mode:DIO" instead of my "mode:DOUT".

blazoncek, 2024-03-23:
Brownout means inadequate PSU for 3.3V.
Not a software issue.

Panossa, 2024-03-23:
> Brownout means inadequate PSU for 3.3V. Not a software issue.

@blazoncek can't be the PSU since I'm using the same setup(s!) I've used before where this error didn't work. Tried two different PSUs earlier, before the problems started. And then the same now.

softhack007, 2024-03-24:
@Panossa I agree with @blazoncek - the message "Brownout detector was triggered" is from ROM and it happens independent from WLED. Yor ESP monitors voltage continuously, and when voltage drops below a critical threshold (even for just a few milliseconds) then the CPU is reset, and you see a reboot message on serial monitor.

You could try a custom build with `-D WLED_DISABLE_BROWNOUT_DET` added to `build_flags`. This switches off the brownout detector completely, which is normally not a good idea but it may help in special scenarios.

On hardware side, adding a big capacitor may help to stabilize the power supply. Also you could use a startup preset with low brightness, to reduce inrush currents at LEDs power-on.

Closing this report,  as WLED cannot do anything more on the software side.

closed 2024-03-24 by softhack007

softhack007, 2024-03-24:
> strips and wired the power and ground in the wrong config

Bad luck  - most likely you toasted your hardware. LED strips are not robust against inverted polarity.
In best case just the first few LEDs are dead (which may explain the constant brownouts), in worst case the whole stripe plus esp32 board is useless now.

softhack007, 2024-03-24:
> Bad luck - most likely you toasted your hardware. LED strips are not robust against inverted polarity. In best case just the first few LEDs are dead (which may explain the constant brownouts), in [...]

Panossa, 2024-03-24:
Just for context: I removed the LED strip after that one time I wired them
wrong. Now I get the brownout error when starting the ESP without anything
attached, as stated above. ^^

And I fear building the software with that ignore flag might end up with me
frying even more, right?