wled/WLED #3847: sporadic restarts

ihavenonick, 2024-03-23:
### What happened?

Since the update to V0.14.2, all of my ESPs (8266 and 32) restart irregularly after at least an hour of standby. Is this intentional or a bug?
I control all of my WLED controllers via IOBroker and have been getting the error message "ECONNRESTE" since the update, which was not the case before.

### To Reproduce Bug

update an ESP 8266 D1 mini or ESP32 S2 to v0.14.2

### Expected Behavior

no restarts

### Install Method

Self-Compiled

### What version of WLED?

v0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

IOBroker Error:
2024-03-23 07:10:06.612 - error: wled.0 (1577122) Websocket connection error : Error: read ECONNRESET

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-23:
Please provide restart crash dumps available on serial interface.
You may also want to obtain debug build to provide more info.

ihavenonick, 2024-03-23:
It affects all of my ESPs with WLED, so the problem should be fairly easy to recreate. I have a default preset that switches the strip off after a reboot, so I noticed that every strip no longer lights up after about 1 hour, or if switched off, flashes very briefly after a reboot.

bouboun59, 2024-03-23:
Hello I have a problem since 14.1 only 14.0 it's ok for me 

WLED
24/02 v14.0 no problem
6/03/2024 20H00  v14.1.B2
10/03/2024  same issue that B1 BUG
10/03/2024 10H15 v14.0 any problem since 12days

I have D1 Mini Nodemcu ESP8266

ihavenonick, 2024-03-23:
All ESP back to v0.14.0 and no problems.

florianru1984, 2024-03-24:
I have the same issue on 2 ESP8266s. ESP32 works fine

mariusstrom, 2024-03-27:
Same issue for me on both 14.2 and 15.1b1. Unfortunately my WLED is in a tricky-to-access location so hard to get crashdumps.

mariusstrom, 2024-03-27:
I have been controlling WLED with the Homebridge [homebridge-wled-ws](https://github.com/smhex/homebridge-wled-ws) plugin to expose it to HomeKit -- after disabling this plugin, things seem to be more stable at least on 15.1b1. Seems that there's something in the plugin and WLED seems to not handle it well.

> Same issue for me on both 14.2 and 15.1b1. Unfortunately my WLED is in a tricky-to-access location so hard to get crashdumps.

blazoncek, 2024-03-27:
ESP8266 *is* limited on RAM. Hammering it with requests (likely `/json`) will bring it down rather soon.

But, no crash dumps with exception decoder, no resolution. That's how it is.

ihavenonick, 2024-03-27:
Not only ESP6266, SEP32S2 mini has the same Problem and more RAM.

I also don't have time to create a dump after every update and document the new bugs. V0.14.0 runs without any problems, that's why I'm sticking with it.
The problem with the crashes occurs with all of my ESP (12 pieces) and is therefore a general bug, which can be reproduced simply by flashing the current WLED bin.

blazoncek, 2024-03-27:
Unfortunately none of my 40+ devices (0.15.0-b1 or newer) exhibit "sporadic restarts".
So, if you do not have the time to create a debug dump, it is ok to close the issue I assume.
Reopen if you feel otherwise.

closed 2024-03-27 by blazoncek

ihavenonick, 2024-03-27:
0.15.0-b1 is not 0.14.2 and 0.14.0. works for me, so you can close and have fun with 0.15.0-b1 ;)