wled/WLED #3852: Stairway-Wipe Usermod Disables Brightness Changes

itanitarek10, 2024-03-24:
### What happened?

Compiled 0.14.2 and can validate the stairway wipe occurs on a boot. However, the brightness can never be modified after compiling with the usermod. By removing the usermod and recompiling, brightness is fine.

Changing the default brightness in LED settings has a minimal impact upon reboot. It does change the brightness, but not as bright as it should be.

### To Reproduce Bug

Add stairway-wipe usermod build flag in platformio.ini
Change userVar0 to a 1 or 2 in wled.h
Compile
Validate that the LEDs light up in a stairway pattern on boot. Rebooting is an easy way to verify as well.
Adjust the brightness slider, see no change.
Change the default brightness in LED settings and see minimal change between brightness levels on boot.

### Expected Behavior

The brightness of the LEDs should be able to change at any point.

Changing the default brightness in LED settings does affect the slider, but that value does not properly translate to the LEDs themselves.

### Install Method

Self-Compiled

### What version of WLED?

0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
None
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-25:
Please contact original author or current maintainer of the usermod. You can find original author either in the accompanying readme document or via Git blame.
WLED team (unless personally motivated) does not maintain usermods. They are user-contributed code.

TMoney1099, 2024-03-25:
Seems like @lost-hope has been maintaining this, at least per the last change to this usermod.

@lost-hope Any luck here?

lost-hope, 2024-03-25:
I dont see a reason for this behaviour, from what i see
I only got it to work again and added buildflags. So not really maintaining it. 
i think @toto79 is the author of the V2 version.

closed 2024-08-01 by github-actions[bot]