wled/WLED #3854: Can not turn on LEDS with version above 14.0

tomlut, 2024-03-25:
### What happened?

I have a Hyperion ambilight set up behind my TV consisting of:

250 LED WS2812 on segment 0, UDP sync interface enabled for main segment
20A 5V power supply (multiple injection points).
Quinled ESP23, data pin GPIO26

Using version 14.0 this **works perfectly**. Syncs with video frames and LEDs on/off control operates as expected.

Using any later version (14.1, 14.2) installed and configured fresh, (to eliminate update json config migration as the cause) the LEDs initially turn on but after turning off the LEDs either via the WLED web interface or the Hyperion interface the LEDs can not be turned back on. 

### To Reproduce Bug

Install 14.1 or later version. 
Turn off LEDs, attempt to turn LEDs back on.

### Expected Behavior

LED control and sync to operate.

### Install Method

Binary from WLED.me

### What version of WLED?

14.1, 14.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Config file:

[wled_cfg_tv-led-strip.json](https://github.com/Aircoookie/WLED/files/14749941/wled_cfg_tv-led-strip.json)

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tomlut, 2024-03-25:
https://github.com/Aircoookie/WLED/issues/3854

willmmiles, 2024-03-29:
OK - I think you can close this as a duplicate of #3720 then.

closed 2024-03-29 by blazoncek