wled/WLED #3858: Missing most effects in custom build

Suxsem, 2024-03-27:
### What happened?

I just compiled a fresh firmware of the 0_15 branch. This is my platform_override:

```
board = esp32dev
platform = ${esp32.platform}
platform_packages = ${esp32.platform_packages}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32}
  -D WLED_RELEASE_NAME=ESP32_audioreactive
  ${esp32.AR_build_flags}
lib_deps = ${esp32.lib_deps}
  ${esp32.AR_lib_deps}
monitor_filters = esp32_exception_decoder
board_build.partitions = ${esp32.default_partitions}
```

Everything works but I only have a bunch of effects, most of them are missing, what I'm doing wrong?

[image]

### To Reproduce Bug

Compile WLED with the posted platform_override

### Expected Behavior

All effects to be there

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0-b2 (build 2403260)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-27:
Some effects have no meaning on non-RGB or single pixel set up.
Such effects are hidden.

closed 2024-03-27 by blazoncek

Suxsem, 2024-03-27:
@blazoncek I selected WS281X but the issue still occurs:

[image]

Suxsem, 2024-03-27:
oohh, it's the number of pixels, nevermind, sorry for bothering!!