wled/WLED #3861: IPv6 support (esp. for mDNS)

T-X, 2024-03-29:
**Is your feature request related to a problem? Please describe.**

I would like to place WLED devices into our Freifunk Wireless Community Mesh Network. And let people in our mesh network play with them. Unfortunately, they won't get discovered through mDNS, as the Gluon firmware filters a lot of multicast messages, to stay more scalable. There now have been new additions to Gluon and the batman-adv mesh routing protocol which will allow using mDNS via IPv6 in these networks again up to a certain network size.

Unfortunately, mDNS over IPv4 very likely won't be usable in the future here though, as RFC4541 ("Considerations for Internet Group Management Protocol (IGMP) and Multicast Listener Discovery (MLD) Snooping Switches"), section [2.1.2, 2)](https://www.rfc-editor.org/rfc/rfc4541#section-2.1.2).  "Data Forwarding Rules" excludes the link-local multicast 224.0.0.0/24 range, which means that IPv4 mDNS with 224.0.0.251 won't be snoopable either.

**Describe the solution you'd like**

If WLED were supporting IPv6 and mDNS over IPv6 then it should be able to answer to an AAAA query on ff02::fb like this anywhere in our mesh:

`[fe80::3e2f:58dc:9f96:b70e]:5353 -> [ff02::fb]:5353 : Standard query 0x0000 AAAA ffhl-wled-schild.local, "QM" question`

And the batman-adv mesh routing protocol would be able to send this request only to hosts participating in mDNS and by that keeping the overhead manageable.

For us, auto-generated link-local IPv6 addresses from the MAC address would be sufficient for now. But of course, it would be great if it could also get global IPv6 addresses via SLAAC (or DHCPv6) or statically configured IPv6 addresses.

**Describe alternatives you've considered**

N/A

**Additional context**

With IPv6 support it would also be easier to allow access over the internet. As we can give each WLED device a unique, publicly accessible IPv6 address, but we can't do the same with IPv4.

Thanks for this great project :-)!

blazoncek, 2024-03-29:
And I just wanted to remove IPv6 stack entirely yesterday to gain some usable RAM and reduce flash size.
BTW WLED does not have any serious security measures or SSL implemented and as such is not suitable to be exposed to the internet.

benley, 2024-06-30:
If WLED had ipv6 support I would use it, but not for exposing it to the internet.  I am gradually switching my private networks to use ipv6 where possible.

That said, if there just isn't enough RAM on an ESP32 to do it, then oh well.  I'd rather have it be reliable :)

RedVyper, 2026-08-29:
@T-X just to let you know that I managed to make ESP32 boards work with Link Local addresses and mDNS. The PR is still WIP but you can test the code yourself.

PR #5822

Moreover later today I also managed to have ULAs and GUAs with SLAAC (and unfortunately only EUI-64 for now).