wled/WLED #3867: Playlist will not go past the 1st Preset

morrisam74, 2024-03-30:
### What happened?

When making a Playlist with defined Presets, the Playlist will only play the 1st Preset in the Playlist and stay on that Preset. 

### To Reproduce Bug

Define at least 2 Presets. Define a Playlist with at least 2 Presets and Test. 

### Expected Behavior

Playlist should go through all the Presets in the Playlist within the timeframe of the Playlist.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-31:
Please provide presets.json and cfg.json.

Mattstir, 2024-04-02:
I came across the same problem today when setting up a new WLED instance with `0.15.0-b1`. I applied config and presets from an identical light project that runs on `0.14.2`. Everything worked as expected, only the transitions in playlist's didn't work for me. 
After downgrading the new instance also to `0.14.2` it started working as expected again without changing anything else.

_Background information in case its interesting:_
Board: D1 mini
LEDs: WS2812B
LED length: 44
Segments 1 or 2 depening on the safed preset 
Happens on different playlists with different durations and different transition times

Ulmerju, 2024-04-02:
See following issues:
#3843 and #3844

morrisam74, 2024-04-04:
Sorry, I was out on Holiday. Attached are the Presets and CFG files. Thank
you for looking into this. I am using a UserMod for servos but I have also
seen this happen for the regular release of 0.15.0-b1.

Tony

Doyle4, 2024-04-07:
Happening to me also, Boot with save and play upon launch, but if a preset is made with multiple effects and set for boot using any method to set it does nothing, will boot up to orange.

Mattstir, 2024-10-22:
This issue seems not fixed! (Please don't close it)
Note: I for my part now always opted for older versions as I don't need the new versions in my projects!

southsko, 2024-12-07:
I had a similar problem where the playlist did nothing.  I flashed back to 14 and it works.  There is a playlist bug

netmindz, 2024-12-08:
Please can you confirm @southsko and @Mattstir if you still see the issue with 0.15.0.rc.1?

southsko, 2024-12-08:
Yes, I created the playlist with  0.15.0.rc.1 and it would not work at all.   I flashed .14 over 0.15.0.rc.1 and it works.

blazoncek, 2024-12-08:
I requested a non-working presets.json file and still, 9 months later, none is provided.

However, my test showed that there might be a bug where 1st item in playlist may not have correct preset saved sometimes. The other issue is that if you save with shuffle option enabled, playlist may be saved with items shuffled.

southsko, 2024-12-08:
{"0":{},"5":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,0,0],[255,255,255],[0,0,0]],"fx":37,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Candy Cane Chase 2"},"10":{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":87,"sx":128,"ix":128,"pal":255,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Glitter RWB","ql":"10"},"1":{"win":"T=1","n":"ON"},"3":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,0,0],[0,0,0],[0,0,0]],"fx":112,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Dancing Shadows","ql":"3"},"2":{"win":"T=0","n":"OFF"},"4":{"on":true,"bri":255,"transition":7,"mainseg":3,"seg":[{"id":0,"start":0,"stop":149,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,0,0],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":1,"start":149,"stop":339,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[0,0,255],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":true,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":2,"start":339,"stop":447,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,170,0],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":3,"start":447,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,170,0],[0,0,0],[0,0,0]],"fx":102,"sx":255,"ix":255,"pal":35,"c1":128,"c2":128,"c3":16,"sel":true,"rev":true,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Fire on house","ql":"4"},"6":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,0,0],[0,0,0],[0,0,0]],"fx":74,"sx":219,"ix":249,"pal":11,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Colortwinkles Rainbow","ql":"6"},"7":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":149,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,0,0],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":1,"start":149,"stop":339,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[0,0,255],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":true,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":2,"start":339,"stop":447,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,170,0],[0,0,0],[0,0,0]],"fx":66,"sx":30,"ix":255,"pal":0,"c1":128,"c2":128,"c3":31,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"id":3,"start":447,"stop":722,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,217,183],[0,0,0],[0,0,0]],"fx":90,"sx" [... cut ...] 0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"Rolling Balls c9"}                                                                                                                                                         ,"18":{"playlist":{"ps":[15,14,16],"dur":[6000,6000,6000],"transition":[7,7,7],"repeat":0,"end":0,"r":0},"on":true,"n":"Xmas Playlist","ql":"18"}         }

blazoncek, 2024-12-09:
Plays well for me.

blazoncek, 2024-12-10:
One more note regarding use of playlist as a boot preset.

I noticed that sometimes 1st preset activated from playlist will not load properly but the playlist timer will count down toward next preset and will correctly apply it (i.e. after 10 minutes from above example). I am guessing a race condition somewhere in `applyPreset()` function or `handlePreset()`.

gbmerrall, 2025-02-12:
New user here and experienced this issue.  Think I've identified the issue. If you select the already selected option then the onChange() event doesn't correctly fire and the playlist (and hence plJson (which forms preset.json) records 0 for the index.

Note to people experiencing this: Just show the playlist editor and manualyl update the the ps list with the numbers of your presets in the web UI. 

Version: 0.15.0 (build 2412100)

To reproduce:
1. Create 3 presets of your choice
2. Create a playlist
3. From the present dropdown select a preset that is *not* the default choice. I'll add preset 1 which is not the default.
4. Open up your browser console and inspect the value of plJson[3] (or whatever your playlist is). You should see the value of the preset recorded

{
    "ps": [
        1
    ],
    "dur": [
        300
    ],
    "transition": [
        7
    ],
    "repeat": 0,
    "end": 0,
    "r": false
}

5. Now add another preset to your playlist but this time just accept the preset that is already selected. 
6. Look at plJson again

{
    "ps": [
        1,
        0
    ],
    "dur": [
        300,
        300
    ],
    "transition": [
        7,
        7
    ],
    "repeat": 0,
    "end": 0,
    "r": false
}

7. Now we'll edit the second entry to specifically select a different present (5 in my case). Again inspect plJson

{
    "ps": [
        1,
        5
    ],
    "dur": [
        300,
        300
    ],
    "transition": [
        7,
        7
    ],
    "repeat": 0,
    "end": 0,
    "r": false
}

I believe this fix here with my somewhat limited JS would be to add a timeout to addPL() or refreshE() to call plePS() which would automatically update plJson with the correct value.

Something along the lines of `setTimeout(() => select.onchange(), 0); `

blazoncek, 2025-02-12:
Workaround: Select non-default (not 1st) preset and then change it to a desired preset.