wled/WLED #3870: plasma effect on random palet devolves into flickering chaos

Matt2D3, 2024-03-31:
### What happened?

after setting the effect to plasma and the color palette to random cycle, it works fine until the next day, where the palette becomes mostly red but flickers to blue occasionally. this happens on esp32 with WLED_0.15.0-b1_ESP32 firmware and can only be resolved by a reboot

### To Reproduce Bug

use esp32 and WLED_0.15.0-b1_ESP32
use random cycle palette 
I have only tested with the plasma effect, but given the changes to palette blending it will most likely happen on other effects

### Expected Behavior

the random cycle palette should be consistent after the first few hours 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.15.0-b1_ESP32.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

this happens on 3 different setups. all use esp32 with WLED_0.15.0-b1_ESP32 firmware, will test esp8266 with equivalent firmware soon

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-03-31:
@dedehai can you take a look?

Matt2D3, 2024-04-03:
it takes around 16h to occour
also I tested it and it does occur on the esp8266.
this bug also persists across effects, so it seems to be an issue with pallet switching specifically (triggered bug through plasma, bug continued when switching to noise 1)

DedeHai, 2024-04-05:
Thanks for finding and reporting this. I wrote up a test and was able to reproduce it now. Problem is an unhandled millis() rollover, causing the palette to fade uncontrollably. It's an easy fix. (PR submitted)

blazoncek, 2024-04-06:
@Matt2D3 please test #3880

Matt2D3, 2024-04-13:
beta 2 is working correctly

closed 2024-06-23 by blazoncek