wled/WLED #3871: v0.15.0-b1 do not sync as source  to other nodes

oltas123, 2024-04-01:
### What happened?

i updated all my nodes to version 0.15.0-b1, after that no sync from any node to the others. After downgrade one node to v14.2 the sync from this node works to all nodes with version 0.15.0-b1 as usual.

### To Reproduce Bug

install version 0.15.0-b1 of the  sync  node

### Expected Behavior

the active sync node should sync the other nodes of the network

### Install Method

Binary from WLED.me

### What version of WLED?

version 0.15.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xrtze, 2024-04-01:
Confirming the same behaviour

blazoncek, 2024-04-01:
It does. You just need to enable sync.
It works a bit different though.

You need to tick notifications.
[image]

xrtze, 2024-04-02:
If I understand correctly, this not enabled on default? Wouldn't it be less missleading, if it was? Bcs rn if you click on the sync button in the top bar of the UI, nothing happens and the feature seems to not work.

blazoncek, 2024-04-02:
This is a **beta** release, not all features are complete and are subject to change at any time.
You are welcome to document changes and submit PR to WLED documentation.

closed 2024-06-23 by blazoncek