wled/WLED #3878: Usermod BH1750 won't compile for ESP8266 (V0.14.2)

BanditJackpot, 2024-04-04:
### What happened?

When trying to custom compile WLED (V0.14.2) for an ESP8266 (Lolin D1 Mini board) with the BH1750 Usermod enabled, WLED will not compile, and i'm presented with **error: statement-expressions are not allowed outside functions nor in template-argument lists**

```
In file included from C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:295,
                 from wled00\wled.h:68,
                 from wled00\usermods_list.cpp:1:
C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:40:34: error: statement-expressions are not allowed outside functions nor in template-argument lists
   40 | #define F(string_literal) (FPSTR(PSTR(string_literal)))
      |                                  ^~~~
C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
wled00\../usermods/BH1750_v2/usermod_BH1750.h:62:31: note: in expansion of macro 'F'
   62 |   String mqttLuminanceTopic = F("");
      |                               ^
*** [.pio\build\nodemcuv2\src\usermods_list.cpp.o] Error 1
``` 

I've followed the usermod readme and ensured all dependencies are included (see below).

I have tried this again with a fresh installation of Visual Studio, Platform.io & a fresh copy of WLED 0.14.2 from Github, but the problem persists.

If I try to compile this usermod for an ESP32 board, it builds successfully, so this seems to be limited to ESP8266 boards only, and I can't figure out why.

### To Reproduce Bug

Compile via Visual Studio, modifying the platformio.ini file to uncomment:
```
default_envs = d1_mini
```

and add the following build flags and library dependencies (as per the readme)

```
[env:d1_mini]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
upload_speed = 921600
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
    -D USERMOD_BH1750
lib_deps = ${esp8266.lib_deps}
    claws/BH1750 @ ^1.2.0
monitor_filters = esp8266_exception_decoder
``` 

### Expected Behavior

When compiling for an ESP32 board, I can compile successfully, so I would expect this to work for an ESP8266 board (and this is also listed in the example for Usermod BH1750

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.2 & 0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
In file included from C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:295,
                 from wled00\wled.h:68,
                 from wled00\usermods_list.cpp:1:
C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:40:34: error: statement-expressions are not allowed outside functions nor in template-argument lists
   40 | #define F(string_literal) (FPSTR(PSTR(string_literal)))
      |                                  ^~~~
C:\Users\PC\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
wled00\../usermods/BH1750_v2/usermod_BH1750.h:62:31: note: in expansion of macro 'F'
   62 |   String mqttLuminanceTopic = F("");
      |                               ^
*** [.pio\build\nodemcuv2\src\usermods_list.cpp.o] Error 1
```
```

### Anything else?

I have searched the forums and discord, and I've only found one other person who has mentioned a similar issue, but with no resolution.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-04:
Please contact original author of the usermod.
WLED team does not maintain usermods unless personally interested.

thepuncake, 2024-04-04:
I'm facing the same issue, should the bh1750 usemod be included in the source code if outdated etc? People will naturally open issues here. Is the author of the particular usermod @softhack007?
Cheers!

closed 2024-04-24 by blazoncek