wled/WLED #3879: Show current CPU Frequency in WebUI and maybe Home Assistant integration

CRCinAU, 2024-04-04:
Now that we have 80 & 160Mhz binaries, it would be good to show which one is actually in use on an ESP8266 based chip.

I recently updated my WLED install on a device I had changed to the 160Mhz binary, but there was no way to tell if it updated to an 80 or 160Mhz version of WLED.

blazoncek, 2024-04-05:
There is no API to detect the actual CPU frequency. If you see no difference and you are getting adequate performance then it doesn't matter what CPU frequency is.

CRCinAU, 2024-04-05:
Yeah - I haven't looked at the WLED code closely - but if its using the normal Arduino libraries, some random code I used on a WebUI is:

```c++
		F("<tr><td>IP Addresses</td><td>") + ip_addr + F("</td></tr>\n") +
		F("<tr><td>MAC Address</td><td>") + WiFi.macAddress().c_str() + F("</td></tr>\n") +
		F("<tr><td>CPU Speed:</td><td>") + ESP.getCpuFreqMHz() + F(" MHz</td></tr>\n") +
		F("<tr><td>Flash Speed:</td><td>") + (ESP.getFlashChipSpeed() / 1000000) + F(" Mhz</td></tr>\n") +
		F("<tr><td>Flash Real Size:</td><td>") + ESP.getFlashChipRealSize() + F(" bytes</td></tr>\n") +
		F("<tr><td>Flash Mode:</td><td>") +
			(
			ESP.getFlashChipMode() == FM_QIO ? "QIO" :
			ESP.getFlashChipMode() == FM_QOUT ? "QOUT" :
			ESP.getFlashChipMode() == FM_DIO ? "DIO" :
			ESP.getFlashChipMode() == FM_DOUT ? "DOUT" :
			"UNKNOWN") + F("</td></tr>\n") +

```

blazoncek, 2024-04-05:
[image]
would that do it?

CRCinAU, 2024-04-05:
That would be beautiful :)

That way, I can see what build has actually been flashed :)

softhack007, 2024-09-28:
availeable since 0.15.0-b2 (8266 and esp32 chips supported)

closed 2024-09-28 by softhack007