wled/WLED #3881: Sync Time if no internet

newwolf, 2024-04-05:
Could it be possible to make WLED sync the internal time from your device time.
Like on the press of a button (sync device time) and it grabs the local device time (from the browser or something?)

Should be very nice when we are at a internetless location, because we use the light as a night/wakeup light with timers for our kid.

blazoncek, 2024-04-05:
It does that when you connect to ESP using phone or PC when NTP is not defined or WLED opens AP.

softhack007, 2024-04-06:
... plus there is the RTC usermod. Just make a  custom build  with RTC usermod, attach a DS3231 or DS1307 module to I2C, and your WLED will take the time/date from the realtime clock chip at startup.