wled/WLED #3882: HTTP request in AP mode (multirelay)

pixelwave, 2024-04-05:
### What happened?

If I connect a WLED ESP32 to a Wifi Network and then try “192.168.178.59/relays?toggle=1,0,1,0” it works!

But it does not seem to work if the WLED ESP32 is in “AP-Mode” (4.3.2.1/relays?toggle=1,0,1,0).

For my final deployment I have the WLED ESP32 not connected to any Wifi network just acting as its own AP.

It seems to be an issue with the “multi-relay” usermod implementation. Because the standard HTTP commands from WLED (like “4.3.2.1/win&A=255”, etc…) work when connected to a Wifi Network as well as in “single AP-Mode”.

### To Reproduce Bug

Put WLED in AP only mode and try to send HTTP request “/relay“ or “/relays?toggle=1,0,1,0“.

### Expected Behavior

Get response / toggle relays.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-05:
Not a bug. This is by design.
Relevant code:
```c++
  /**
     * connected() is called every time the WiFi is (re)connected
     * Use it to initialize network interfaces
     */
    inline void connected() override { InitHtmlAPIHandle(); }
```

blazoncek, 2024-04-05:
JSON API

pixelwave, 2024-04-06:
Ok the JSON API seems to work even when in AP-mode. Now I need to rewrite the remote controller script for JSON commands. While this is possible previously with my HTTP API approach I was relaying on the status request command to update the remote controller UI "Status Request: http://[device-ip]/relays". 

This command does not seem to be available in the JSON API (https://mm.kno.wled.ge/usermods/MultiRelay/).

I still do not quite understand the "design decision" to not allow HTTP requests from the usermod, but standard HTTP requests, JSON requests etc are allow in AP-mode?

Feels more like a bug then a "feature" ...

blazoncek, 2024-04-06:
If you are using MM then open an issue there. Also you can, as already said, implement the functionality yourself.
I have no intention of doing so.

pixelwave, 2024-04-06:
> If you are using MM then open an issue there. Also you can, as already said, implement the functionality yourself. I have no intention of doing so.

What is "MM"? 

> Also you can, as already said, implement the functionality yourself.

As stated before I am no expert. I am usually trying to cobble together some code and am happy if stuff works. ;) Everything worked fine in this project until I switched to AP-only mode and encountered that - for my understanding - weird behaviour.

> I have no intention of doing so.

Also I am not pushing you to implement anything. I only asked if it is maybe a simple line comment / uncomment in the current code. But when you mentioned it being a deliberate "design decision" I was just trying to understand the logic behind. Because it is a decision that imo makes no sense at all - but maybe I am missing something ...

closed 2024-08-30 by github-actions[bot]