wled/WLED #3884: V0.15.0-b1 causes WRGB PWM strip to not shutdown completely

gaaat98, 2024-04-05:
### What happened?

After updating to 0.15.0-b1 an RGBW PWM strip will leave one channel extremely dim instead of powering off, only an update to the color wheel or the sliders will turn the strip completely off.

The issue is not present in WLED 0.14.3. I would like to provide other details but I don't know what could be helpful..

### To Reproduce Bug

1. Turn on the strip
2. Set to Solid Color
3. Select a color from the color wheel
4. Power off the strip: the "predominant" channel of the selected color will remain on albeit very dim
5. Select a color on the color wheel/move one of the sliders: the strip will power off completely

### Expected Behavior

The led strip should completely power off

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b1 (build 2403190)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-06:
Can you please describe in more detail including which colors are selected and what sequence of actions you followed.
Please also include your configuration (cfg.json).

After you press Power button and LEDs turn off after the transition (default 0.7s) there are no more updates to LEDs and changing selected color has no effect on LEDs themselves.

gaaat98, 2024-04-06:
The steps to reproduce the bug have been performed immediately after flashing the WLED_0.15.0-b1_ESP02.bin.gz via OTA. Apart from random colors on the color wheel I tested the predefined colors on the left side of the WebUI. The issue appears with every color except for black which does not turn on anything.

The exact workflow I followed for testing was:
1. Update from 0.14.3 to 0.15.0-b1
2. With strip off select a color from UI, make sure that the the first slider below the color wheel is at max, the issue doesn't seem to be present if the slider is set to a lower value
3. Turn on
4. Turn off (either via webui, android app or physical button mapped to the toggle api)
5. Verify that the strip is dim but still on (if the color was purple then red and blue channels will be still on etc., also the white channel will be on if its slider was set to max)
6. Select again a color (either the same or a different color) from one of the UI (can be different from the UI used in 2)
7. Verify that the strip is fully off
(7b. Tap again the color of 6: the strip will turn on again as it was in 5 (on but very dim))
9. Repeat from 2 with another color

I do not exclude that the issue could be related to my particular hardware setup since I have one RGB analog strip connected to the RGB mosfets and a separate white analog strip controlled by the W channel mosfet, but it is strange since with 0.14.x this did not happen.

Here's my [wled_cfg.json](https://github.com/Aircoookie/WLED/files/14893227/wled_cfg.json)

blazoncek, 2024-04-06:
Try, and report:
- set normal PWM frequency (now you have slow)
- set transition time to 1s or more
- set transition time to 0

PWM handling was modified in 0.15 to include 12bit PWM resolution and better low brightness level handling.

gaaat98, 2024-04-06:
- Set Clock to Normal and Fastest, issue persists
- Set transition time to 1000ms and 2000ms, issue persists
- Set transition time to 0ms, issue persists

I also noticed that if I touched a color right after saving settings the strip will turn on at full brightness, while the UI reports it as being off... Not sure if that's related or not

blazoncek, 2024-04-06:
Please contact me on Discord. I have a possible fix, but needs to be tested.

PR #3887 (2024-04-06): fix for #3884
That 0.5f offset causes pwmBri to never be 0 even if _bri is set to 0 so the led won't turn off completely. Removing it fixes #3884.
closed 2024-04-06 by blazoncek