wled/WLED #3888: Usermod TetrisAI compile error

robertvb83, 2024-04-06:
### What happened?

I am not able to compile WLED 0.15 with -D Usermod_TetrisAI

### To Reproduce Bug

compile latest 0.15 branch with VS PlatformIO

with override -D USERMOD_TETRISAI

### Expected Behavior

TetrisAI should compile :-)

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
In file included from wled00/../usermods/TetrisAI_v2/tetrisaigame.h:22:0,
                 from wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:7,
                 from wled00/usermods_list.cpp:217:  
wled00/../usermods/TetrisAI_v2/tetrisai.h: At global scope:
wled00/../usermods/TetrisAI_v2/tetrisai.h:221:10: error: 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)' cannot be overloaded
     bool findBestMoveNonBlocking(GridBW grid, std::vector<Piece>::iterator start, std::vector<Piece>::iterator end, Rating* bestRating)
          ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:206:10: error: with 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)'
     bool findBestMoveNonBlocking(GridBW grid, std::vector<Piece>::iterator start, std::vector<Piece>::iterator end, Rating* bestRating)
          ^
wled00/../usermods/TetrisAI_v2/tetrisai.h: In member function 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)':
wled00/../usermods/TetrisAI_v2/tetrisai.h:211:17: error: 'expression' was not declared in this scope      
         switch (expression)
                 ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:213:18: error: 'INIT' was not declared in this scope
             case INIT:
                  ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:219:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from wled00/usermods_list.cpp:217:0:
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h: In function 'uint16_t mode_2DTetrisAI()':
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: error: no match for 'operator!' (operand type is 'TetrisAIGame')
   if (!tetrisai_data->tetris || (tetrisai_data->tetris.nLookAhead != nLookAhead
       ^
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: note: candidate: operator!(bool) <built-in>   
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: note:   no known conversion for argument 1 from 'TetrisAIGame' to 'bool'
*** [.pio\build\custom_build\src\usermods_list.cpp.o] Error 1
============ [FAILED] Took 8.00 seconds ============

Environment    Status    Duration
-------------  --------  ------------
custom_build   FAILED    00:00:08.000
======= 1 failed, 0 succeeded in 00:00:08.000 =======
PS D:\WLED\WLED-0_15> pio run -e custom_buildpio run -e custom_build
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-07:
Please contact (tag) original author of the usermod.

robertvb83, 2024-04-07:
Hallo @muebau please have a look

muebau, 2024-04-07:
thank you. I will have a look in that but I think some of the newer code sneaked in there and it should be enough to simply remove it.

muebau, 2024-04-12:
I found the original commit (first working version). I remember there was some problem if the effect was changed often but it is the version I created the PR for.
There is a problem to compile it with 0.15b2 at the moment.

How would I fix the code? Should I create another PR?

blazoncek, 2024-04-12:
> How would I fix the code? Should I create another PR?

Yes.

muebau, 2024-04-12:
@robertvb83 
I created a version that compiles on a branch:
https://github.com/muebau/WLED/tree/fix_usermod_tetisai

Please use this and I will create a pull request later.

robertvb83, 2024-04-12:
I did a quick check in github actions (i am only on my mobile phone)

`In file included from wled00/../usermods/TetrisAI_v2/tetrisaigame.h:22:0,
                 from wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:7,
                 from wled00/usermods_list.cpp:213:
wled00/../usermods/TetrisAI_v2/tetrisai.h: At global scope:
wled00/../usermods/TetrisAI_v2/tetrisai.h:221:10: error: 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)' cannot be overloaded
     bool findBestMoveNonBlocking(GridBW grid, std::vector<Piece>::iterator start, std::vector<Piece>::iterator end, Rating* bestRating)
          ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:206:10: error: with 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)'
     bool findBestMoveNonBlocking(GridBW grid, std::vector<Piece>::iterator start, std::vector<Piece>::iterator end, Rating* bestRating)
          ^
wled00/../usermods/TetrisAI_v2/tetrisai.h: In member function 'bool TetrisAI::findBestMoveNonBlocking(GridBW, std::vector<Piece>::iterator, std::vector<Piece>::iterator, Rating*)':
wled00/../usermods/TetrisAI_v2/tetrisai.h:211:17: error: 'expression' was not declared in this scope
         switch (expression)
                 ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:213:18: error: 'INIT' was not declared in this scope
             case INIT:
                  ^
wled00/../usermods/TetrisAI_v2/tetrisai.h:219:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
In file included from wled00/usermods_list.cpp:213:0:
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h: In function 'uint16_t mode_2DTetrisAI()':
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: error: no match for 'operator!' (operand type is 'TetrisAIGame')
   if (!tetrisai_data->tetris || (tetrisai_data->tetris.nLookAhead != nLookAhead
       ^
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: note: candidate: operator!(bool) <built-in>
wled00/../usermods/TetrisAI_v2/usermod_v2_tetrisai.h:119:7: note:   no known conversion for argument 1 from 'TetrisAIGame' to 'bool'
*** [.pio/build/custom_build/src/usermods_list.cpp.o] Error 1
========================== [FAILED] Took 6.61 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
custom_build   FAILED    00:00:06.610
==================== 1 failed, 0 succeeded in 00:00:06.610 ====================
Error: Process completed with exit code 1.`

blazoncek, 2024-04-12:
FYI
```
git remote add muebau https://github.com/muebau/WLED.git
git checkout -b --no-track muebau/fix_usermod_tetisai
git pull
```
and then compile.

robertvb83, 2024-04-12:
https://github.com/Aircoookie/WLED/assets/36081149/76d89502-9b76-43f3-93ca-e5cb4a154593

It works and is super cool! Unfortunately i cannot get AR, Animartrix and Tetris on one build due to file size exceed error

@muebau Thanks for providing such a cool usermod and thanks for the quick fix!

robertvb83, 2024-04-12:
[image]

blazoncek, 2024-04-12:
> It works and is super cool! Unfortunately i cannot get AR, Animartrix and Tetris on one build due to file size exceed error

Get ESP with 8MB flash or modify partitions,

muebau, 2024-04-12:
I created a new PR https://github.com/Aircoookie/WLED/pull/3897

robertvb83, 2024-04-13:
> > It works and is super cool! Unfortunately i cannot get AR, Animartrix and Tetris on one build due to file size exceed error
> 
> Get ESP with 8MB flash or modify partitions,

Thanks for the hint. I found a partition file at MoonModules that works. Animartrix, AR, TetrisAI on one 4MP dev board! (Still deactivate IR and mqtt).

[WLED_ESP32_4MB_256KB_FS.csv](https://github.com/MoonModules/WLED/blob/mdev/tools/WLED_ESP32_4MB_256KB_FS.csv)

closed 2024-06-23 by blazoncek