wled/WLED #3889: Applying incorrect 2D settings will make it crash

DedeHai, 2024-04-07:
### What happened?

saving incorrect 2D settings while an FX is running causes a crash and settings are not saved.

### To Reproduce Bug

setup one 2D panel. I used 256 leds, one output, 1 panel, 16x16.
load Akemi FX, save as preset and apply at boot. then reboot.
without changing LED outputs, go to 2D settings, set to two panels, 16x16 each, set the second one to x-offset=16 (offset is important, I suspect something tries to write to the non-existing leds outside the first 16x16 panel)
apply settings and it does crash.
happens on ESP32 and ESP32 S2 and S3. 
was not able to reproduce this on 0.14.3

### Expected Behavior

should not crash or at least save the settings.

### Install Method

Self-Compiled

### What version of WLED?

0.15 (latest at 7. 4. 2024)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
WS FileRead: /index.htm
JSON buffer locked. (17)
JSON buffer size: 8023 for request: 6
JSON content length: 5417
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 4807 for request: 8
JSON content length: 2190
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 2498 for request: 3
JSON content length: 1360
JSON buffer released. (17)
WS FileRead: /skin.css
Not-Found HTTP call: /presets.json
WS FileRead: /presets.json
WS client connected.
JSON buffer locked. (12)
JSON buffer size: 2494 for WS request (1360).
heap 184964
Sending WS data to a single client.
JSON buffer released. (12)
WS client disconnected.
settings resp0
settings resp10
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008d356  PS      : 0x00060833  A0      : 0x8008f35e  A1      : 0x3ffdc730  
A2      : 0xffffffff  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060820  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffdc730  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x3ffdc94c  
A14     : 0x3f407a19  A15     : 0x00000037  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffffffff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d353:0x3ffdc730 0x4008f35b:0x3ffdc760 0x4008f9a1:0x3ffdc780 0x40085f2d:0x3ffdc7a0 0x40085f5e:0x3ffdc7c0 0x4008637d:0x3ffdc7e0 0x4000beaf:0x3ffdc800 0x401b55af:0x3ffdc820 0x4011f9d5:0x3ffdc8a0 0x401ba372:0x3ffdc8e0 0x400e8c62:0x3ffdc900 0x4010d01d:0x3ffdc990 0x4011c241:0x3ffdcdd0 0x4011c899:0x3ffdce70 0x40130b6d:0x3ffdce90 0x4012e5b9:0x3ffdcee0 0x4012e641:0x3ffdcf20 0x401243ad:0x3ffdcf40 0x4012442d:0x3ffdcf80 0x40124a5e:0x3ffdcfa0 0x4008c486:0x3ffdcfd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c

---WLED 0.15.0-b2 2404050 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ6 rev.0, 2 core(s), 240MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 40MHz.
heap 312096
JSON buffer allocated: 32767
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
heap 279260
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 16 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 256) with type 22 and pins 16,255 (itype 23). mA=55/850
PIN ALLOC: Pin 0 successfully allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
heap 271876
Initializing strip
Loading custom palettes
Loading custom ledmaps
Matrix ledmap:
 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271,
 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287,
 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303,
 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319,
 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333, 334, 335,
 336, 337, 338, 339, 340, 341, 342, 343, 344, 345, 346, 347, 348, 349, 350, 351,
 352, 353, 354, 355, 356, 357, 358, 359, 360, 361, 362, 363, 364, 365, 366, 367,
 368, 369, 370, 371, 372, 373, 374, 375, 376, 377, 378, 379, 380, 381, 382, 383,
 384, 385, 386, 387, 388, 389, 390, 391, 392, 393, 394, 395, 396, 397, 398, 399,
 400, 401, 402, 403, 404, 405, 406, 407, 408, 409, 410, 411, 412, 413, 414, 415,
 416, 417, 418, 419, 420, 421, 422, 423, 424, 425, 426, 427, 428, 429, 430, 431,
 432, 433, 434, 435, 436, 437, 438, 439, 440, 441, 442, 443, 444, 445, 446, 447,
 448, 449, 450, 451, 452, 453, 454, 455, 456, 457, 458, 459, 460, 461, 462, 463,
 464, 465, 466, 467, 468, 469, 470, 471, 472, 473, 474, 475, 476, 477, 478, 479,
 480, 481, 482, 483, 484, 485, 486, 487, 488, 489, 490, 491, 492, 493, 494, 495,
 496, 497, 498, 499, 500, 501, 502, 503, 504, 505, 506, 507, 508, 509, 510, 511,
Playlist unloaded.
Request to apply preset: 1
heap 271144
Usermods setup
heap 271144
Defaulf WiFi used.
initServer
heap 216672
Initial connect or forced reconnect.
Defaulf WiFi used.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-07:
I'm sure you can use exception decoder. 😉 
It will save me a  lot of time.

DedeHai, 2024-04-08:
does this help?

settings resp10
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008d3b6  PS      : 0x00060033  A0      : 0x8008f216  A1      : 0x3ffdbec0  
A2      : 0xffffffff  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060020  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffdbec0
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060020  A13     : 0x3f428100  
A14     : 0x3f409c24  A15     : 0x0000000d  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffffffff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: 0000000000000000

Backtrace: 0x4008d3b3:0x3ffdbec0 0x4008f213:0x3ffdbef0 0x4008f859:0x3ffdbf10 0x40085fd1:0x3ffdbf30 0x40086002:0x3ffdbf50 0x40086421:0x3ffdbf70 0x4000beaf:0x3ffdbf90 0x401e7f17:0x3ffdbfb0 0x4015149f:0x3ffdc030 0x401ed14d:0x3ffdc070 0x400f378c:0x3ffdc090 0x4013bbfd:0x3ffdc190 0x40146e87:0x3ffdc870 0x40147419:0x3ffdc910 0x40146723:0x3ffdc930 0x40166361:0x3ffdc960 0x40162e69:0x3ffdc9a0 0x40162f05:0x3ffdc9f0 0x40157665:0x3ffdca10 0x401576b5:0x3ffdca40 0x401576f9:0x3ffdca60 0x401577ea:0x3ffdca80 0x40157894:0x3ffdcaa0 0x4008c536:0x3ffdcad0
  #0  0x4008d3b3:0x3ffdbec0 in uxPortCompareSet at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #1  0x4008f213:0x3ffdbef0 in multi_heap_internal_lock at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #2  0x4008f859:0x3ffdbf10 in multi_heap_malloc at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305
  #3  0x40085fd1:0x3ffdbf30 in heap_caps_malloc at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #4  0x40086002:0x3ffdbf50 in heap_caps_malloc_default at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #5  0x40086421:0x3ffdbf70 in _malloc_r at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
  #6  0x4000beaf:0x3ffdbf90 in ?? ??:0
  #7  0x401e7f17:0x3ffdbfb0 in VFSImpl::open(char const*, char const*) at C:/Users/dedehai/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/libraries/FS/src/vfs_api.cpp:31
  #8  0x4015149f:0x3ffdc030 in LITTLEFSImpl::exists(char const*) at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/LittleFS.cpp:105
  #9  0x401ed14d:0x3ffdc070 in fs::FS::exists(char const*) at C:/Users/dedehai/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/libraries/FS/src/FS.cpp:258
  #10 0x400f378c:0x3ffdc090 in WS2812FX::setUpMatrix() at wled00/FX_2Dfcn.cpp:86
  #11 0x4013bbfd:0x3ffdc190 in handleSettingsSet(AsyncWebServerRequest*, unsigned char) at wled00/set.cpp:756
  #12 0x40146e87:0x3ffdc870 in serveSettings(AsyncWebServerRequest*, bool) at wled00/wled_server.cpp:578 (discriminator 5)
  #13 0x40147419:0x3ffdc910 in std::_Function_handler<void (AsyncWebServerRequest*), initServer()::{lambda(AsyncWebServerRequest*)#9}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at wled00/wled_server.cpp:274
      (inlined by) _M_invoke at c:\users\dedehai\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #14 0x40146723:0x3ffdc930 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at c:\users\dedehai\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:2271
  #15 0x40166361:0x3ffdc960 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at .pio/libdeps/esp32dev_debug/ESPAsyncWebServerWLED/src/WebServer.cpp:71
  #16 0x40162e69:0x3ffdc9a0 in AsyncWebServerRequest::_onData(void*, unsigned int) at .pio/libdeps/esp32dev_debug/ESPAsyncWebServerWLED/src/WebRequest.cpp:1015
  #17 0x40162f05:0x3ffdc9f0 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#5}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio/libdeps/esp32dev_debug/ESPAsyncWebServerWLED/src/WebRequest.cpp:1015
      (inlined by) _M_invoke at c:\users\dedehai\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #18 0x40157665:0x3ffdca10 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at .pio/libdeps/esp32dev_debug/AsyncTCP@src-39de97abf7348c44d4dda815b8aab0ae/src/AsyncTCP.cpp:1076       
  #19 0x401576b5:0x3ffdca40 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32dev_debug/AsyncTCP@src-39de97abf7348c44d4dda815b8aab0ae/src/AsyncTCP.cpp:1076
  #20 0x401576f9:0x3ffdca60 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32dev_debug/AsyncTCP@src-39de97abf7348c44d4dda815b8aab0ae/src/AsyncTCP.cpp:1076
[... cut ...]
  #23 0x4008c536:0x3ffdcad0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c

---WLED 0.15.0-b2 2404050 INIT---

blazoncek, 2024-04-08:
It happens in LittleFS while checking if `/2d-gaps.json` exists.
Can't do much about it.

Perhaps you ran out of memory?

DedeHai, 2024-04-08:
the crash reason at the beginning mentions 'StoreProhibited'. definitely not out of memory, this is on an ESP32 with just 256 LEDs enabled. Here I set it to 2D with 2x 16x16 panel. that works fine, even with just one output defined. the crash happens, when I set the offset of the second panel to 16, which probably writes to a non-existing buffer or out of range of the LED buffer.

blazoncek, 2024-04-08:
There might be an issue when resizing a matrix as buffer is already allocated from the boot sequence.
It still baffles me, why the crash in `FS::exists()`.

blazoncek, 2024-04-08:
Try this in `FX_2Dfcn.cpp`line ~68
```c++
    if (customMappingTable) delete[] customMappingTable;
    customMappingTable = new uint16_t[getLengthTotal()];
```

scaytrase, 2024-04-08:
same here. saving 2d settings with 16x16 bottom right serpentime causes my esp-wroom-32 to crash immediately and do not save settings.

if there is a bin file  with fix I can try it

DedeHai, 2024-04-09:
I tried with the changes in `FX_2Dfcn.cpp` it does not help at all.

I found out (with this fix only? did not check) the crash also happens when having it on 'solid' color and adding a second panel, with no offset (so overlapping panels). After the crash (no crashdump was created unfortunately) the wifi settings are lost (no connections available):
this is the output after the crash:

```
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 0 successfully allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
heap 279012
Initializing strip
No busses, init default
PIN ALLOC: Pin 16 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 30) with type 22 and pins 16,255 (itype 23). mA=55/850
Loading custom palettes
Loading custom ledmaps
heap 277988
Usermods setup
heap 277988
Defaulf WiFi used.
initServer
heap 223708
Initial connect or forced reconnect.
Defaulf WiFi used.
initConnection() called.
No connection configured.
Opening access point WLED-AP
```

so seems like the config gets corrupted. After reboot, it starts the AP but it does not save any settings, in fact, it just crashes.
When opening 4.3.2.1 to configure wifi, it outputs this:

```
Not-Found HTTP call: /chrome-variations/seed
Captive portal
settings resp1
Not-Found HTTP call: /connecttest.txt
Captive portal
Not-Found HTTP call: /connecttest.txt
Captive portal
```

the settings page loads without issues but when saving the wifi settings, this happens:

```
Writing settings to /wsec.json...
JSON buffer locked. (4)
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x4008c23b  PS      : 0x00060236  A0      : 0x8008c3d0  A1      : 0x3ffb0070
A2      : 0x3ffbb080  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc23d4  A8      : 0x8008d249  A9      : 0x3ffb0080
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000008  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008c238:0x3ffb0070 0x4008c3cd:0x3ffb00b0 0x401d6fad:0x3ffb00d0 0x4008928b:0x3ffb00f0 0x40089795:0x3ffb0110 0x401d6f3a:0x3ffb0180 0x401528e5:0x3ffb01b0 0x4014d03f:0x3ffb01d0 0x4014dbc9:0x3ffb0200 0x40151b95:0x3ffb0290 0x4015262c:0x3ffb02f0 0x4015276d:0x3ffb0340 0x4014c54d:0x3ffb0360 0x401d9861:0x3ffb0490 0x4000bcc5:0x3ffb04b0 0x401e5679:0x3ffb04d0 0x401e3db2:0x3ffb04f0 0x4014ba05:0x3ffb0570 0x4014baa4:0x3ffb05a0 0x40107af2:0x3ffb05c0 0x4011016d:0x3ffb0780 0x401402e5:0x3ffb1f50 0x40140bdb:0x3ffb1f90 0x4016aadd:0x3ffb1fb0 0x4008c526:0x3ffb1fd0
  #0  0x4008c238:0x3ffb0070 in xQueueGenericReceive at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2053
  #1  0x4008c3cd:0x3ffb00b0 in xQueueTakeMutexRecursive at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2053
  #2  0x401d6fad:0x3ffb00d0 in spi_flash_op_lock at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:56
  #3  0x4008928b:0x3ffb00f0 in spi_flash_disable_interrupts_caches_and_other_cpu at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:97
  #4  0x40089795:0x3ffb0110 in spi_flash_guard_start at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:165
      (inlined by) spi_flash_read at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:530
  #5  0x401d6f3a:0x3ffb0180 in esp_partition_read at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:313
  #6  0x401528e5:0x3ffb01b0 in littlefs_api_read at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/littlefs_api.c:22
  #7  0x4014d03f:0x3ffb01d0 in lfs_bd_read at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/lfs.c:3162
  #8  0x4014dbc9:0x3ffb0200 in lfs_dir_fetchmatch at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/lfs.c:3162
  #9  0x40151b95:0x3ffb0290 in lfs_dir_find at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/lfs.c:3162
  #10 0x4015262c:0x3ffb02f0 in lfs_rawstat at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/lfs.c:3162
  #11 0x4015276d:0x3ffb0340 in lfs_stat at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/lfs.c:3162
  #12 0x4014c54d:0x3ffb0360 in vfs_littlefs_stat at .pio/libdeps/esp32dev_debug/LittleFS_esp32/src/esp_littlefs.c:272
  #13 0x401d9861:0x3ffb0490 in esp_vfs_stat at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:734 (discriminator 4)
  #14 0x4000bcc5:0x3ffb04b0 in ?? ??:0
  #15 0x401e5679:0x3ffb04d0 in stat at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/syscalls/../../../.././newlib/libc/syscalls/sysstat.c:12
  #16 0x401e3db2:0x3ffb04f0 in VFSImpl::open(char const*, char const*) at C:/Users/Max Power/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/libraries/FS/src/vfs_api.cpp:41
  #17 0x4014ba05:0x3ffb0570 in fs::FS::open(char const*, char const*) at C:/Users/Max Power/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/libraries/FS/src/FS.cpp:258
[... cut ...]
  #21 0x401402e5:0x3ffb1f50 in WLED::loop() at wled00/wled.cpp:202 (discriminator 1)
  #22 0x40140bdb:0x3ffb1f90 in loop() at C:/Users/Max Power/GIT/WLED/wled00/wled00.ino:20
  #23 0x4016aadd:0x3ffb1fb0 in loopTask(void*) at C:/Users/Max Power/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #24 0x4008c526:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

blazoncek, 2024-04-09:
Please read the exception decoder. The crash happens within file system library. Not much can be done about that apart from filing a bug report in ESP platform.

I am guessing that you are using 80MHz flash clock which your flash chip may not like. Please lower it to 40MHz and try again.

I did fix allocation bug when resizing matrix though.

DedeHai, 2024-04-09:
Your alloacation fix seems to have resolved this, I cannot get it to crash anymore.
Regarding file system being to blame: if a bad allocation messes with memory, the stack backtrack is not always useful. A runaway pointer is ofthen hard to pin down as it can change values in places you do not expect, causing random crashes at random places.

closed 2024-04-09 by blazoncek