wled/WLED #3891: Flickering on 6th segment on WS2805 BTF strip after 30 brightness

LydellMoore, 2024-04-08:
### What happened?

I've noted on my setup and testing with B2 and WS2805 strips from BTF that any brighter than 30 brightness there is distinct flickering on the 6th segment on multiple cut strips (all 7 segments long) but NOT on strips that are 6 segments long.  I've run the strip[s in multiple series (ie different series of 6/7 segment strips( and this always affects the 7 segments but starts to affect the 6 segment (on segment 5 aka one before the end) when 80% bright on any effect.

### To Reproduce Bug

Turn strips up over 30 brightness in any effect setting.

### Expected Behavior

no flickering at all

### Install Method

Binary from WLED.me

### What version of WLED?

WLED V .15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sansillusion, 2024-04-09:
Sounds like a possible power delivery issue. I might be wrong but I would
make sure the strips are getting enough amperage/voltage to run reliably.

blazoncek, 2024-04-09:
@LydellMoore please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Please also consult [KB](https://kno.wled.ge).

closed 2024-06-23 by blazoncek