wled/WLED #3893: LoLin NodeMCU V3 ESP8266 not functioning after installing latest version

vietanhdang, 2024-04-11:
### What happened?

Hello,
I'm encountering an issue when trying to install the latest version of WLED on the LoLin NodeMCU V3 ESP8266. After installation, my device is not functioning properly and I am unable to access it through the web interface. Below are some detailed information about the issue I'm facing:

### To Reproduce Bug

Install the latest version of WLED on the LoLin NodeMCU V3 ESP8266.
Power on the device.
Attempt to access the web interface using the device's IP address.

### Expected Behavior

The device should function properly after installing the latest version of WLED.
Able to access the web interface of the device via the IP address.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I have tried reinstalling and rebooting the device multiple times but still encounter this issue.
I have checked and ensured that the device's network connection is stable.
I would greatly appreciate any assistance from the community in resolving this issue. Thank you!

Best regards,
Viet Anh Dang

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-11:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2024-04-11 by blazoncek

vietanhdang, 2024-04-13:
> Hello, I have the same issue on the same board. Could you solve your problem?

As a workaround, I've installed version v.0.14 until a proper fix is available.