wled/WLED #3896: LED Custom mA/LED is not saved

robertvb83, 2024-04-12:
### What happened?

When setting mA/LED to Custom, the value of mA is set to 55 and other values are not saved

### To Reproduce Bug

Set mA/LED to Custom, change mA to 30 and hit save...

### Expected Behavior

Should save custom valies

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-12:
Thanks for reporting. Value is saved correctly and is used correctly by ABL, the error is in display. You can verify by looking at cfg.json
Will fix it in next release.

robertvb83, 2024-05-04:
Works as it should, thanks for fixing

closed 2024-05-04 by robertvb83