wled/WLED #3900: 0.15.0-b2(&b1): Palette effect randomly resetting and changing direction

Plawasan, 2024-04-13:
### What happened?

After upgrading to 0.15.0-b2 (audioreactive) and applying the palette effect to either of the two segments on my TV I get a random reset of the effect every ~2s - seems to also be changing direction, see video.

Downgrading to 0.14.3 fixes the issue, it seems to be also present in 0.15.0-b1

Running on Athom 2CH controller.

### To Reproduce Bug

Apply Palette effect to a segment

### Expected Behavior

no resets

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b2 (build 2404100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-13:
Hi. Did you read release notes? It is clearly stated that Palette effect has been changed and that is a **BREAKING** change. It means that default behaviour has changed and you will need to update effect parameters to get the look you were used to.

The change to the effect is obvious when used with 2D set up.

To get the old behaviour, use the following:
- set the Shift slider to get desired speed
- set the Rotation slider to 128
- uncheck Animate rotation
- you may want to experiment with Animate shift and Shift slider

closed 2024-04-13 by blazoncek

Plawasan, 2024-04-13:
I routinely do, but must have missed this one as I originally skipped b1.. thanks for the pointers.