wled/WLED #3906: Consider using Framebuffer::GFX and Adafruit's NeoMatrix Mapping for 2D support

marcmerlin, 2024-04-15:
I can't re-open https://github.com/Aircoookie/WLED/issues/2197 but in the last 3 years since I opened it, I've heard of more peole using wled and now trying to do 2D with it, which it wasn't originally built to do.

as discussed on https://www.facebook.com/groups/LEDSAREAWESOME/posts/2360328277495951/?comment_id=2362375507291228&reply_comment_id=2363141040548008&notif_id=1713103541286836&notif_t=group_comment_mention  
I pointed out that adafruit's mapping work is both super important and useful when you chain matrices as getting coordinate right, is non trivial:
[image]

Maybe it would be a good time to re-visit looking at
- http://marc.merlins.org/perso/arduino/post_2020-03-16_Framebuffer_GFX_-Choosing-between-its-3-2D-APIs_-FastLED-XY_-NeoMatrix_-and-LEDMatrix_-and-detail-of-its-many-supported-hardware-backends.html
- https://github.com/marcmerlin/Framebuffer_GFX
and especially
- https://github.com/marcmerlin/FastLED_NeoMatrix  (which is built on top of framebuffer::GFX)

top easily add 2D framebuffer support that will work with all kinds of hardware, support the weird Z mappings of chained panels, and allow using code that uses FastLED, Adafruit::GFX, and LEDMatrix primitives and APIs.  All this stuff is already written and already exists, might as well re-use it :)

marcmerlin, 2024-04-15:
Cc @embedded-creations and I'll reference this on the FB leds are awesome discussion

marcmerlin, 2024-04-15:
To add context, I would have contributed to WLED, but I wrote most of my code before WLED had enough to do what I needed (or actually barely existed), so I have WLED like support on esp32 and other chips, plus the full 2D matrix and API support, but it wasn't meant to be as easy to use by othres as WLED , so it makes more sense for WLED to continue to grow and expand with its user base, than for me to add the WLED bits missing to my code.
https://github.com/marcmerlin/NeoMatrix-FastLED-IR

the middle of that page shows the wifi support which is custom to my need, but can be adapted of course
https://marc.merlins.org/perso/arduino/post_2022-01-09_LED-Pants-and-Shirt-v5-on-ESP32-and-Raspberry-Pi-with-P2-RGBPanels-and-Wifi.html

All that said, I still think most people will be better off using wled in the long term once equivalent 2D support can be added (by that I mean the APis I mentioned and neomatrix support)

blazoncek, 2024-04-16:
You are welcome to make a PR. The same requirements from #2420 are still valid and have to be respected though.

FYI there are now about 190 effects that may need rewriting when changing low-level APIs. Most of these are not 2D. And BTW WLED's canvas is called `Segment`. It has its own coordinates which are Cartesian TL-BR oriented (Y axis inverted) which ten map to the outside world via `WS2812FX` class and its LED mapping.

The example from the picture above is fully possible with current automatic mapping built into WLED.

marcmerlin, 2024-04-16:
Ok, I was given incorrect information when I asked/looked. If WLED already fully supports complex inter and intra zmapping of LED within and between LED matrices, equivalent to what Neomatrix and LEDMatrix offer, then adding support for FastLED::NeoMatrix isn't as useful. 
I don't personally need this in WLED since I wrote my own before WLED had any 2D support (but my WIFI support is much limited compared to WLED's, so I send people to WLED if they need wifi and only need addressible pixel support), but I wanted to make sure it did have that complex 2D mapping support and if it already does it's all good.
If anything is missing that my libs have and WLED doesn't, someone who needs that support can hopefully take my relevant code and merge it in, although it won't be be since I don't have the need myself for WLED at this point (it would take over 100h of work to port all my code to WLED at this time, and it doesn't support the multi 2D API support I have and that allows me to use other 2D hardware like TFTs or RGBPanels, or 2D and arduino emulation straight on linux for quicker code writing and debugging.
But I do welcome anyone taking that code and making it more available to others.

closed 2024-06-23 by blazoncek