wled/WLED #3909: Crashes with v14 when set to PWM RGB

agmorpheus, 2024-04-16:
### What happened?

I am a new user and have now also been working with the ESP8266 (Wemos D1) together with WLED. I use WLED with a PWM RGB LED strip.

I have made the following observation. 
Installed version 14.3 and set the setup to PWM RGB accordingly. When I change the colors, I very often have crashes.
I did some testing and the crashes did not occur when WS281x is set. Then I can change the colors at will.
I then tested this immediately with version 14.0. Unfortunately I also get the crashes there.
Only version 13 runs stable under PWM RGB without crashes.

I have already seen that there are probably a lot of crashes with the current version at the moment. I have noticed this especially with PWM RGB.

Translated with DeepL.com (free version)

### To Reproduce Bug

Set PWM RGB and change the colors a lot

### Expected Behavior

No crash when changing color

### Install Method

Binary from WLED.me

### What version of WLED?

14.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-16:
Please get a debug build, connect USB and provide a crash dump.
If you can it would be even better if you could compile yourself and get a crash dump using exception decoder.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support obtaining debug builds or compile instructions.

agmorpheus, 2024-04-16:
I build the recent WLED version now for wemos d1 and flashed it. Now it is working.

blazoncek, 2024-04-17:
If it keeps working without crashes then please close the issue.

closed 2024-04-17 by agmorpheus