wled/WLED #3911: Starting Sunrise effect in a segment after having a brightness of 0 flashes at full brightness one time

distante, 2024-04-17:
### What happened?

when starting sunrise configured for a segment, coming from an state where the brightness was 0 (hole strip), the segment flashes at full brightness and then the effect starts.

### To Reproduce Bug

go from
```
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"","col":[[0,0,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":40,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

to
```
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":300,"grp":1,"spc":0,"of":0,"on":false,"frz":false,"bri":255,"cct":127,"set":0,"n":"All","col":[[255,0,0],[0,0,0],[0,0,0]],"fx":106,"sx":64,"ix":170,"pal":40,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":50,"stop":160,"grp":1,"spc":0,"of":0,"on":false,"frz":false,"bri":188,"cct":127,"set":0,"n":"Luz de lectura ","col":[[255,169,86],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":2,"start":0,"stop":49,"grp":1,"spc":0,"of":0,"on":false,"frz":false,"bri":71,"cct":127,"set":0,"n":"Pre lectura","col":[[140,93,0],[0,0,0],[0,0,0]],"fx":104,"sx":1,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":false,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":3,"start":161,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":69,"cct":127,"set":0,"n":"Post lectura ","col":[[255,133,0],[0,0,0],[0,0,0]],"fx":104,"sx":19,"ix":59,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

### Expected Behavior

The sunrise effect should start without flashing

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.3 (build 2404040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-17:
FYI both presets posted above have state "on" and brightness 255.

The first preset has Solid effect and color black, the second has red color set. When switching from black to red the LEDs will transition from completely off to red color. Since the difference between 0 and 1 (or more) is quite visible, you get the impression of "flash".
To verify, extend the transition time from 0.7s to several seconds and it will be more obvious. If you want to avoid "flash" set the color in second preset to black as well.

distante, 2024-04-17:
I see, the transition is the main culprit. If I put the sunrise effect with `transition: 0` no flash happens.

I still it can be considered a bug, or at least something for the documentation?

I discovered this problem because at night time, I put the Tinkleup effect with a magenta color and then I just turn the strip off. 

```json
{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"","col":[[255,0,0],[0,0,0],[0,0,0]],"fx":106,"sx":64,"ix":170,"pal":40,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

Then in the morning, home assistant loads the sunrise effect and the flash was always there. I just used the defaults presets and the default value in the config is `0.7s` for transitions.

blazoncek, 2024-04-17:
It is definitely not a bug. 😄 
If you find documentation lacking you can always make contribution to WLED-Docs.

distante, 2024-04-17:
Already there :)

closed 2024-04-17 by distante