wled/WLED #3912: Постоянное зависание прошивки

SergioLuxx, 2024-04-17:
### What happened?

All versions after 0.13.3 hang the board tightly until the power is completely lost.
I use python code that sends commands to the lamp via websockets about changing the mode and its parameters, which are selected randomly. Sending occurs at a random time from the list, every [0.2, 0.3, 0.4, 0.5, 1.0] seconds.
I am seriously upset because I am forced to sit on the stable working 0.13.3, which handles such a load absolutely correctly. Every time you release a new version, I install it to check, with the hope that everything will finally work, but no! My lamp starts to freeze constantly after 5 - 10 seconds of running the code to change modes. The LEDs either turn off completely and the board no longer responds to anything, there is not even ping, because it is disconnected from wifi, or random LEDs light up and continue to light, still showing no signs of life. After de-energizing, the lamp works again for 5 - 10 seconds and freezes again. 
And constant websocket requests (0.5) seconds json.loads(ws.recv())["state"]["on"] to check the current state of the lamp do not work at all. Again, unlike version 0.13.3. On all versions after - empty lines or False are returned. when the lamp actually works.
Please fix this already.
Controller used - ESP32
https://youtube.com/shorts/7GvTXqnoz9k?feature=share python waits for the lamp to appear on the network, after which it starts sending commands - an instant freeze occurs.
https://youtube.com/shorts/BL8JGxSuO_I?feature=share complete freeze after a short time of operation.
https://youtube.com/shorts/8ri8UbO42uM?feature=share automatic reboot, after a short time of operation, after which it completely freezes.
https://youtube.com/shorts/1mHL42oSASI?feature=share works absolutely correctly on version 0.13.3. On this firmware the lamp can work without interruption.

### To Reproduce Bug

Randomly select the palette number and its characteristics, send these changes via websockets to the lamp with a random delay of [0.2, 0.3, 0.4, 0.5, 1.0] seconds.
Request the lamp for its current state via an active websocket connection every 0.5 seconds.

### Expected Behavior

Correct operation with quick mode changes, as in version 0.13.3. Transmitting correct readings upon request json.loads(ws.recv())["state"]["on"], as in version 0.13.3.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-05-01:
Hi, why would you want to change effects 5-10 times per second? Are your eyes even fast enough to recognise the running effect? If you want something like a strobe, we have dedicated effects for this....

blazoncek, 2024-05-01:
That's not a bug but technical limitation of a microcontroller which cannot process as many requests in such a short amount of time given all the other tasks it has to perform.
I agree that this could be mitigated but that the task for a library WLED uses, not WLED itself.

Please provide a debug output and possible crash dump using exception decoder.

If you are unsatisfied then, please, stick with 0.13 which has a bit less to do unfortunately.

samvel-galstyan, 2024-07-16:
This causes problems when connected to Home Assistant. Every day I have to unplug and plug again the esp32. I cant even ping to it. My router shows that it was disconnected. I am not changing anything so fast. But perhaps HA is sending some commands which cannot be handled by ESP/WLED

blazoncek, 2024-07-16:
Please provide Wireshark packet capture and WLED debug output.
You will need to upload firmware with debug enabled and monitor serial interface.

closed 2024-11-21 by github-actions[bot]