wled/WLED #3915: Can not perform manual update v0.14.0->v0.14.3 failed

vitalii-dmytruk, 2024-04-18:
### What happened?

Manual update of v0.14.0 (build 2310130) can not be updated to v0.14.3 (ESP8266)

### To Reproduce Bug

1. download binary from [https://github.com/Aircoookie/WLED/releases/download/v0.14.3/WLED_0.14.3_ESP8266.bin](https://github.com/Aircoookie/WLED/releases/download/v0.14.3/WLED_0.14.3_ESP8266.bin)

2. disable OTA update lock in security settings

3. Update manually using downloaded binary

### Expected Behavior

Successful update

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 (build 2310130)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Don't have any
```

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-04-19:
This could be a known issue, as the firmware for 8266 with 2MB flash is over the limit of what's possible for OTA. Unfortunately that's a limitation affecting 8266 boards with less than 4MB flash.

As a workaround, you could try OTA with the `.bin.gz` file from the release page.

* https://github.com/Aircoookie/WLED/issues/3257

vitalii-dmytruk, 2024-04-19:
> From the file system size displayed above, this is a 1MB device. Those
> can't be updated OTA.

Got it. Thanks for response

closed 2024-04-19 by Aircoookie

vitalii-dmytruk, 2024-04-20:
I found the following setting in the LED Preferences. It's possible I have
4KB memory. Is it possible there is another reason why update doesn't work?