wled/WLED #3916: Detect if no changes (JSON interface) received for a period of time?

pjapk, 2024-04-18:
I have a use case where I'm displaying LED colours/effects based on outside info, sent to WLED via Node-Red approximately every 5 mins. Occasionally, my data source is offline so I have a "data offline indicator" preset that makes it obvious that the data is out of date. However, I've realised that if WLED itself disconnects from WiFi for any reason, then I have no means of sending a preset command to indicate this.

What I think would be useful is a means of WLED detecting when it was last changed/updated, and if the last update was x mins ago, "do something" (such as set another appropriate indicator preset). 

For context, I'm basically displaying an approximation (rounded to nearest integer) of blood glucose levels for Type 1 diabetes, with a rising/falling slightly/quickly effect, with 5-LED segments of different colours to make it easy to see from a distance, the difference between, say, 5 & 15 without having to try to squint/count LEDs, but if WIFI goes offline, I currently can't indicate that the numbers presented are stale due to loss of comms to WLED.

Hopefully this is clear - it's only my second ever feature request on Github! :-D

(Also, thanks to everyone involved for an amazing tool!)

willmmiles, 2024-04-19:
The general term for this feature would be a "dead man switch".  This could be done by a quick hack to the 'auto save' usermod, replacing the [`saveSettings()` call](https://github.com/Aircoookie/WLED/blob/57665e896400a43f8cf2a71d975db418739c2c63/usermods/usermod_v2_auto_save/usermod_v2_auto_save.h#L152) with a call to `applyPreset()`.

blazoncek, 2024-04-19:
The easiest way to achieve this is by writing a usermod. Perhaps modifying AutoSave as suggested.

pjapk, 2024-04-19:
Thanks for the comments guys, makes sense, & Dead Mans Switch is exactly what I'm looking for though I'm no coder by any stretch so will go see what I can hack!