wled/WLED #3917: DDP & Art-Net Matrix  Serpentine Check

dosipod, 2024-04-19:
### What happened?

When using DDP and Art-Net , the master and receiver  serpentine option is mismatched hence in case the receiver  is serpentine  then transmission from master will only work correctly if serpentine  is unchecked or if the receiver  is set as 1D strip .

This has been noticed since a long time even with earlier versions but might be could be fixed in 15 as had a chance to start to test in more details prior to mass migration in test environments  . 

This is a minor issue when using two wled units to send and receive but issue becomes complicated as that would also apply to external apps like jinx and xlights  when used with larger or more complicated fixtures 

### To Reproduce Bug

Setup a matrix on receiver   with serpentine  and also with serpentine  on the master 

### Expected Behavior

Serpentine option should match on sender and receiver  

### Install Method

Self-Compiled

### What version of WLED?

15.0-b3  as of today and all other forks inculding MM

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
NA
```

### Anything else?

Take it easy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-19:
You have an option to disrespect ledmaps when receiving live data since #3554 in 0.15. It will also disable internal ledmaps.

closed 2024-04-19 by blazoncek

dosipod, 2024-04-19:
It is working