wled/WLED #3918: New Wipe Effect for stairs = "WipeStairs"

aLAN-LDZ, 2024-04-19:
**Is your feature request related to a problem? Please describe.**
It's not a problem, but I'm missing an effect that I wanted to write myself, but unfortunately I'm having a little problem with it.

**Describe the solution you'd like**
I would like to create a new effect based on the wipe effect that will be useful as a basic effect for the staircase. The effect would turn on individual LEDs just like the wipe effect. Then he left the LEDs on for some time. Then he turned off the LEDs individually like a wipe, only in the opposite direction

**Additional context**
The easiest way to explain it is with this piece of code:

```
for ( int i = 0; i < SEGLEN; i++)
{
  SEGMENT.setPixelColor(index, WHITE);
  delay(x)
}
delay(10000)
for ( int i = SEGLEN; i > 0; i--)
{
  SEGMENT.setPixelColor(index, BLACK);
  delay(x)
}
```

So far, I have managed to write an effect that lights up all the LEDs. I'm missing the next two steps:

1) leaving the LEDs on for some time
2) turning the LEDs off in the reverse order of turning them on

```
uint16_t stairs(void)
{
  uint32_t mainColor = SEGCOLOR(0);
  uint32_t secondColor = SEGCOLOR(1);

  SEGMENT.fill(secondColor);

  uint32_t cycleTime = 750 + (255 - SEGMENT.speed)*150;
  uint32_t perc = strip.now % cycleTime;
  uint16_t prog = (perc * 65535) / cycleTime;
  uint16_t ledIndex = (prog * SEGLEN) >> 15;

  for (int i = 0; i < SEGLEN; i++)
  {
    if (i < ledIndex)
    {
      SEGMENT.setPixelColor(i, mainColor);
    }
  }

  return FRAMETIME;
}
static const char _data_FX_MODE_STAIRS[] PROGMEM = "Stairs@!,!;!,!;!";
```

This seems to me quite an interesting idea for an effect that can be used for stairs. If you can, help me finish it. Thank you for any advice

aLAN-LDZ, 2024-04-19:
Are you sure? because when I looked at it, I only saw whole segments lighting up at once, not wipe style

aLAN-LDZ, 2024-04-21:
I managed to write working code. This could probably be improved further.

```//StairsEffect
uint16_t stairs(void)
{
  uint32_t mainColor = SEGCOLOR(0);
  uint32_t secondColor = SEGCOLOR(1);

  uint32_t cycleTime = 750 + (255 - SEGMENT.speed)*150;
  uint32_t perc = strip.now % cycleTime;
  uint16_t prog = (perc * 65535) / cycleTime;
  
  ///// DOBRY KOD ///////

  uint16_t ledIndex = 0;
  uint16_t ledIndex2 = 0;

  if (((prog * SEGLEN) >> 15) < SEGLEN)
  {
    ledIndex = (prog * SEGLEN) >> 15;
  }
  else
  {
    ledIndex = SEGLEN - 1;
  }

  ledIndex2 = (SEGLEN - 1) - ledIndex;

  uint32_t delayTime = 5000;
  uint32_t startTime = 0;

  if (SEGMENT.step == 0)
  {
    //Serial.print("led: ");
    //Serial.println(ledIndex);
    SEGMENT.setPixelColor(ledIndex, mainColor);

    if (ledIndex == SEGLEN - 1)
    {
      SEGMENT.step = 1;
      startTime = strip.now;
      Serial.print("Step: ");
      Serial.println(SEGMENT.step);
    }

  }

  if (SEGMENT.step == 1)
  {
    if (strip.now - startTime >= delayTime)
    {
      SEGMENT.step = 2;
      Serial.print("Step: ");
      Serial.println(SEGMENT.step);
    }

  }

  if (SEGMENT.step == 2)
  {

    if (ledIndex2 != 0 && SEGMENT.aux0 == 0)
    {
      SEGMENT.aux0 = 1;
    }

    if (SEGMENT.aux0 == 1)
    {
      //Serial.print("led2: ");
      //Serial.println(ledIndex2);
      SEGMENT.setPixelColor(ledIndex2, secondColor);
    }

    if (ledIndex2 == 0 && SEGMENT.aux0 == 1)
    {
      SEGMENT.step = 3;
      SEGMENT.aux0 = 0;

      Serial.print("aux: ");
      Serial.println(SEGMENT.aux0 );
      Serial.print("step: ");
      Serial.println(SEGMENT.step );
    }   

  } 
  ///// KONIEC /////////

  /////// TEST ////////
  if (SEGMENT.step == 3)
  {
      if(ledIndex == 0)
      {
        SEGMENT.step = 0;
        Serial.print("step: ");
        Serial.println(SEGMENT.step );
      }
  }

  return FRAMETIME;
}
static const char _data_FX_MODE_STAIRS[] PROGMEM = "Stairs@!,!;!,!;!";
//StairsEffectEnd```

shoelzer, 2024-07-05:
@aLAN-LDZ `Stairs@!,!;!,!;!` is [effect metadata](https://kno.wled.ge/interfaces/json-api/#effect-metadata)

closed 2024-11-11 by github-actions[bot]