wled/WLED #3922: GPIO 5 does not work on 0.15.0-b2

bandi13, 2024-04-22:
### What happened?

Simply updating to 0.15.0-b2 makes one of my panels no longer light up. I use GPIO 5 for panel 9 in a 2D matrix. Reverting back to 0.14.3 (or even 0.15-0-b1) fixes the issue. I use nine WS2812B panels in a 3x3 matrix. The connections are soldered on to a circuit board, so rewiring won't be the easiest. I use the WLED_N_ESP32.bin type for the firmware.

### To Reproduce Bug

Ugrade the firmware to 0.15.0-b2. Completely reproducible by doing a "Manual OTA". The 9th panel stays lit with whatever was drawn on it last.

### Expected Behavior

I expect the same configuration to work regardless of the version the firmware is running.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b2 (build 2404100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

My suspicion is that some new feature is mapped onto GPIO5, but the release notes don't mention anything.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-22:
There are no restrictions or default assignments for GPIO5. Please do a factory reset and configure the device from scratch, it may be a configuration issue.
You can backup configuration and presets from "Security & Updates" settings page.

If resetting the device has no effect on behaviour then please get a debug build (from @srg74 repository) and post boot sequence serial output. You will need to attach the device to computer using USB cable and use tool for dumping serial output (like Putty or other terminal tool).

blazoncek, 2024-04-22:
You need a debug build.

blazoncek, 2024-04-22:
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Dev/0.15.0-b2/Debug_builds

bandi13, 2024-04-22:
Whoops, sorry, it was really late for me. Here's the output with the debug
image:
>>>
WLED RESET
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

---WLED 0.15.0-b2 2403280 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 40MHz.
heap 302524
JSON buffer allocated: 32767
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
Registering usermods ...
heap 269512
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 27 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 256) with type 22 and pins 27,255 (itype
23). mA=55/555

            PIN ALLOC: Pin 23 successfully allocated by 0x82 (130)
Successfully inited strip 1 (len 256) with type 22 and pins 23,255 (itype
21). mA=55/555

            PIN ALLOC: Pin 2 successfully allocated by 0x82 (130)
Successfully inited strip 2 (len 256) with type 22 and pins 2,255 (itype
21). mA=55/555

           PIN ALLOC: Pin 26 successfully allocated by 0x82 (130)
Successfully inited strip 3 (len 256) with type 22 and pins 26,255 (itype
21). mA=55/555

            PIN ALLOC: Pin 22 successfully allocated by 0x82 (130)
Successfully inited strip 4 (len 256) with type 22 and pins 22,255 (itype
21). mA=55/555

            PIN ALLOC: Pin 4 successfully allocated by 0x82 (130)
Successfully inited strip 5 (len 256) with type 22 and pins 4,255 (itype
21). mA=55/555

           PIN ALLOC: Pin 25 successfully allocated by 0x82 (130)
Successfully inited strip 6 (len 256) with type 22 and pins 25,255 (itype
21). mA=55/555

            PIN ALLOC: Pin 21 successfully allocated by 0x82 (130)
Successfully inited strip 7 (len 256) with type 22 and pins 21,255 (itype
21). mA=55/555

            PIN ALLOC: Pin 5 successfully allocated by 0x82 (130)
Successfully inited strip 8 (len 256) with type 22 and pins 5,255 (itype
21). mA=55/555

[... 146 lines cut ...]

          Slow strip 41/23.

                           Loop took 43ms.

                                          Usermods took 0ms.

                                                            Strip took 42ms.

Slow strip 40/23.

                 Loop took 42ms.

                                Usermods took 0ms.

                                                  Strip took 41ms.

                                                                  Slow
strip 40/23.
<<<

Looks like there's a different "successful" code (0x85 vs 0x82) on the
faulty GPIO. Also, I'm not sure about those "Slow strip" errors.

-András

blazoncek, 2024-04-22:
You are using Audioreactive usermod which prohibits more than 8 buses/outputs on versions prior to 0.15.0-b3.
Please use (unreleased ATM) 0.15.0-b3.

bandi13, 2024-04-22:
I just used the version you linked for the Debug version. Previously I was
loading
https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/WLED_0.15.0-b2_ESP32.bin
which exhibits the issue. I don't think that's the Audioreactive version.

-András

softhack007, 2024-04-22:
> I just used the version you linked for the Debug version. Previously I was loading https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/WLED_0.15.0-b2_ESP32.bin which exhibits the issue. I don't think that's the Audioreactive version.

I agree with the conclusion from @blazoncek - this line in the log
> AR: Failed to initialize sound input driver. Please check input PIN

means you're running a firmware that includes the audioreactive usermod. In version WLED_0.15.0-b2 (or older 0.15 pre-releases), WLED with audioreactive will not drive any LEDs on output 9 or 10. Technically its a resource conflict on the I2S driver - outputs 9 & 10 need the I2S unit, however the unit is already "claimed" by audioreactive even when the usermod is disabled.

bandi13, 2024-04-22:
Note, there's a
https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/WLED_0.15.0-b2_ESP32_audioreactive.bin
version in addition to the one I used. I'm using a release version which
exhibits the issue. Either there's an issue with the releases having
Audioreactive in the versions that aren't supposed to have it, or it's not
an Audioreactive issue at all.

bandi13, 2024-04-24:
I tried on the latest 0.15 build from source as well, and the 9th panel does not work. I will try to put an oscilloscope on to the line. My thinking is maybe the timing isn't correct on it? This week I'm pretty busy, but I'll try to get to it one evening.

blazoncek, 2024-04-24:
You can compile custom binary by switching to parallel I2S. Almost all necessary code exists in bus_wrapper.h. You will need to select I2S for the first 8 buses in such case.

Konos3, 2024-04-24:
I have the same issue with another data pin GPIO27, downloaded from https://install.wled.me/, not audioreactive and 9 led strips

blazoncek, 2024-04-25:
@Konos3 please provide cfg.json and debug dump.

bandi13, 2024-04-26:
I ended up hooking up an oscilloscope, and on my 9th panel (GPIO5) I don't get any output with the official https://github.com/Aircoookie/WLED/releases/download/v0.15.0-b2/WLED_0.15.0-b2_ESP32.bin release.
My cfg.json is here: [wled_cfg.json](https://github.com/Aircoookie/WLED/files/15124512/wled_cfg.json). Though I don't think it's relevant since other versions work fine with the same configuration. I'll try to get you a debug dump when I can.

blazoncek, 2024-04-26:
FYI found the bug. It was well hidden.

closed 2024-10-27 by blazoncek