wled/WLED #3927: Unable to set static Ethernet IP address

JonnyTech, 2024-04-24:
### What happened?

Trying to configure the Ethernet card on my ESP32-POE-ISO to have a fixed IP address does not work. It always obtains one from the DHCP server instead.

### To Reproduce Bug

Go to http://<wled-ip>/settings/wifi and set an IP address
(alternatively send the relevant string via JSON API)
Wait or reboot and expect the new IP address to work

[image]

### Expected Behavior

Static IP address should be available on the Ethernet port

### Install Method

Binary from WLED.me

### What version of WLED?

Installed version:WLED 0.15.0-b2 (ESP32-D0WDQ5 build 2404100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Linked discussion on Discourse: https://wled.discourse.group/t/set-ethernet-ip-address-via-api/11683/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-04-25:
@JonnyTech I seen you on the forum and was just wondering you did the same test with  Static gateway added also . This should be your router IP address , just to clarify you tested that  . I have ordered some Ethernet boards to test such cases but sadly it will be a while until they would arrive

JonnyTech, 2024-04-25:
Thank you very much @dosipod  - it does indeed require a gateway address, even though I do not have nor need one, the gateway must still be set with a non-zero value. Your help, on both platforms, is very much appreciated.

closed 2024-04-25 by JonnyTech