wled/WLED #3928: Issue while multiple data outputs

Konos3, 2024-04-24:
### What happened?

After installing the latest beta version (0.15.0-b2) one of the 9 outputs that I was using (D27 and later D26 to test) on my ESP32 stopped working, I reinstalled the same version and it did not get fixed. In my case I was able to get it to work again by downgrading to the official release.

### To Reproduce Bug

Use 9 data pins (9 led strips), for me it happens with D27 and D26 (didn't test any other pins other than 26)

### Expected Behavior

All of the LED strips/outputs to work.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I can reproduce it if you want, just send me a message on discord Konos3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Konos3, 2024-04-24:
Same as #3922

blazoncek, 2024-04-25:
closed as a duplicate

closed 2024-04-25 by blazoncek