wled/WLED #3931: E1.31 input is overiden by web UI brightness slider and power button

xhpohanka, 2024-04-26:
### What happened?

If I use sACN input I would need it to work with maximum priority. Unfortunately I noticed even if UI shows that it receives E1.31 the leds are off or on brightness smaller then expected. Then I found that in such state the power button and brightness slider still works and probably have precedence over E1.31.
It is confusing and I sometimes found my device with brightess slider set to 0 after cold restart. Then I need to go to web UI and move it to full to get E1.31 working.

### To Reproduce Bug

- setup E1.31
- start E1.31 transmitter
- check web ui showing that the WLED is receivng E1.31 live data
- check brightness slider and on off button

### Expected Behavior

brightness slider has no impact when E1.31 is active
on off button has no impact when E1.31 is active

### Install Method

Binary from WLED.me

### What version of WLED?

2403190

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-26:
There is a setting for maximum brightness in Sync settings.

xhpohanka, 2024-04-26:
> There is a setting for maximum brightness in Sync settings.

Thank you, I have missed this settings. When `Force max brightness` is checked it works as I expect...

closed 2024-04-26 by xhpohanka