wled/WLED #3932: is there a physical Button that can change presets ?

sci666-2023, 2024-04-26:
if not can it be added to any datapin ?   

so if the button is pressed it changed to the next saved preset ?

or can several buttons can be added and every button can be mapped to a specific preset ?

example:
Button 1 = normal condition  -> LED Preset 1: white light
Button 2 = yellow allert -> LED Preset 2: yellow blinking leds
Button 3 = red alert -> LED Preset 3: red blinking leds

blazoncek, 2024-04-26:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
You can also familiarize yourself with [KB](https://kno.wled.ge).

closed 2024-04-26 by blazoncek

sci666-2023, 2024-04-26:
mh i dont need help :), your rev page does not show anything that it is button controlled so i think its not implemented now ...

softhack007, 2024-04-26:
Start reading here https://kno.wled.ge/features/macros/#buttons

robertvb83, 2024-05-04:
> mh i dont need help :), your rev page does not show anything that it is button controlled so i think its not implemented now ...

I had the exact same question :-) https://github.com/Aircoookie/WLED/issues/3787

What you want is actually already available and is described here in the presets section
https://kno.wled.ge/features/presets/

Scroll down to the last entry of the table. You only need to define some presets and then create a special cycling preset. This preset is then assigned to your button.

[image]