wled/WLED #3934: problema installazione wled

dipietrofilippo, 2024-04-26:
### What happened?

sto provando a caricare [WLED-HexClock](https://github.com/Bulduper/WLED-HexClock.git) , ma sto avendo molti problemi.
vi invio uno screenshot dell'errore che si presenta quando cerco di caricare il repository su Visual Studio Code
[image]

### To Reproduce Bug

non riesco a installare il software su esp8266, mi crea molti errori su platformio e non riesco a proseguire il progetto

### Expected Behavior

PackageException: Package version 2.2.1+sha.6d2540d doesn't satisfy requirements ~2.0.7 based on PackageMetadata <type=library name=ESPAsyncWebServerWLED 
version=2.2.1+sha.6d2540d spec={'owner': None, 'id': None, 'name': 'ESPAsyncWebServer', 'requirements': '~2.0.7', 'uri': 'git+https://github.com/Aircoookie/ESPAsyncWebServer.git'}

### Install Method

Binary from WLED.me

### What version of WLED?

wled aircookie

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-26:
English please.

dipietrofilippo, 2024-04-26:
I'm trying to load WLED-HexClock, but I'm having a lot of problems.
I send you a screenshot of the error that appears when I try to upload the repository to Visual Studio Code

closed 2024-04-26 by blazoncek