wled/WLED #3936: esp32-s3-zero can not boot

namezys, 2024-04-27:
### What happened?

I'm trying to flash to a small board from Ali with a 4Mb flash.
At first, I tried to use master and config for the 8Mb board but got the same error: header has information that it is 8Mb version

### To Reproduce Bug

just build and flash

### Expected Behavior

can boot

### Install Method

Self-Compiled

### What version of WLED?

0_15

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
ELF file SHA256: 4f3d44314a8826c1

E (298) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403792c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (270) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:326 (flash_ret == ESP_OK)

Backtrace: 0x403796b2:0x3fceb180 0x4037f329:0x3fceb1a0 0x40385d6d:0x3fceb1c0 0x42072dfe:0x3fceb2f0 0x40379021:0x3fceb320 0x403cd757:0x3fceb350 0x403cda16:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

  #0  0x403796b2:0x3fceb180 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037f329:0x3fceb1a0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40385d6d:0x3fceb1c0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x42072dfe:0x3fceb2f0 in do_core_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:318
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:433
  #4  0x40379021:0x3fceb320 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:632
  #5  0x403cd757:0x3fceb350 in ?? ??:0
  #6  0x403cda16:0x3fceb380 in ?? ??:0
  #7  0x403c9929:0x3fceb4b0 in ?? ??:0
  #8  0x40045c01:0x3fceb570 in ?? ??:0
```

### Anything else?

As I can see it tries to flash as 8Mb
```
Processing esp32s3_4M_qspi (board: esp32-s3-devkitc-1; platform: espressif32@ ~6.3.2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
npm run build

> wled@0.15.0-b3 build
> node tools/cdata.js

          ##  ##      ##        ######    ######
        ##      ##    ##      ##        ##  ##  ##
        ##  ##  ##  ##        ######        ##  ##
        ##  ##  ##  ##        ##            ##  ##
          ##  ##      ######    ######    ######
                 build script for web UI

Web UI is already built
(node:60469) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-28:
S3 with 4MB flash is supported in 0.15.0-b3.
Flashing it correctly is still a bit tricky and requires esptool ATM.
There are instructions (and binary files) available on Discord.

closed 2024-04-28 by blazoncek

jonny190, 2024-11-20:
I've tried following the instructions but cannot get a successful flash, i end up in a boot loop with the below 

```
Using 'COM34' as serial port.
Showing logs:
[18:35:23]ESP-ROM:esp32s3-20210327
[18:35:23]Build:Mar 27 2021
[18:35:23]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:35:23]Saved PC:0x4037b09d
[18:35:23]SPIWP:0xee
[18:35:23]mode:DIO, clock div:1
[18:35:23]load:0x3c100020,len:0x4aa9c
[18:35:23]load:0x3fc977b0,len:0x5554
[18:35:23]load:0x42000020,len:0xfaac0
[18:35:24]load:0x3fc9cd04,len:0x11b4
[18:35:24]load:0x40374000,len:0x137b0
[18:35:24]SHA-256 comparison failed:
[18:35:24]Calculated: c563a9f1d2649eb831dc7de7e66adc2204ed0d7f01416d9468391bd1422825a9
[18:35:24]Expected: fcd8e48e667305f8be370639c0489fe21bcf9b2248341b97155e9417bb141f15
[18:35:24]Attempting to boot anyway...
[18:35:24]entry 0x40378c14
[18:35:24]ESP-ROM:esp32s3-20210327
[18:35:24]Build:Mar 27 2021
[18:35:24]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:35:24]Saved PC:0x4037b08d
[18:35:24]SPIWP:0xee
[18:35:24]mode:DIO, clock div:1
[18:35:24]load:0x3c100020,len:0x4aa9c
[18:35:25]load:0x3fc977b0,len:0x5554
[18:35:25]load:0x42000020,len:0xfaac0
[18:35:25]
Serial port closed!
```

Any more pointers

blazoncek, 2024-11-20:
@jonny190 You're not flashing correctly.
For simplicity use PIO and choose correct build environment.

Some pointers: https://wled.discourse.group/t/cant-get-s2-mini-to-flash/10410/3

Or just use unofficial web flasher by @wled-install

jonny190, 2024-11-20:
I was getting boot issues using the web flasher... but PIO works weirdly