wled/WLED #3937: Audio reactive effects for PWM RGB Strips

gaaat98, 2024-04-27:
Hi! I was trying to replace some cheap TUYA RGB PWM controllers (specifically [this](https://aliexpress.com/item/1005001621143902.html)) with Audioreactive WLED, but after spending the evening wiring everything up I realized that the audioreactive effects are only available for addressable strips (as also confirmed by #3494).

My question/feature request now is: it's possible to implement some simple effects such as changing brightness/color/effect speed based on sound? The cheap Tuya controller did it, and I think also WLED should be able to do it!

If I were to implement these effects myself, where should I start?

Thanks in advance

dosipod, 2024-04-28:
@gaaat98  I do not use any analog strips but a  trick we used for a long time to test is just create fake w281x output ( even setup a matrix if you like 2D audio effects ) . If you do that then you would have access to those effects in ui . Not exactly sure how good those effects would be on your setup but I just tested that with esp32 onboard led and it kinda works with audio effects. Better then nothing :)

gaaat98, 2024-04-28:
> @gaaat98 I do not use any analog strips but a trick we used for a long time to test is just create fake w281x output ( even setup a matrix if you like 2D audio effects ) . If you do that then you [...]

Thanks, this seems to be working also with my analog strips! I wonder if there's a way to turn this workaround into an actual feature

blazoncek, 2024-04-28:
If effect works with a single pixel it should expose capability using metadata. That's all.

gaaat98, 2024-04-29:
So would it be welcome if I updated the metadata of the sound reactive effects that still do something "cool" on single pixels despite being designed for addressable strips or is it better not to?

blazoncek, 2024-04-29:
Just be conservative and it will be ok. 😉

closed 2024-05-07 by gaaat98