wled/WLED #3938: Add swagger

RobThree, 2024-04-28:
It would be really cool if the [json api](https://kno.wled.ge/interfaces/json-api/) also served a (static) [swagger.json](https://swagger.io/) file which would make 'discovering' and more importantly implementing the API a lot simpler (automatic client generation on most plaforms and languages). It also serves as (machine readable) documentation. Adding a (also static) [swagger-UI](https://petstore.swagger.io/) would be extra bonuspoints.

Basic steps:

* [ ] Write the `swagger.json`
* [ ] handle the `/json/swagger.json` request by serving the static file
* [ ] (Optional) Also serve the required (static) swagger-ui files

Ofcourse, this would mean that when changes are made to the api this document needs to be kept in sync. But I do really thing it would be worth the effort.

I could (help with) write the `swagger.json` file, but I'm not sure on the other two steps. If anyone is interested in helping or teaming up I would love to prepare a PR. But if there's no interest then I won't make all these efforts in the first place. So: is there interest?

rodericj, 2025-01-28:
I would also love this. I've not really done cpp development but I'm happy to take a stab at it

rodericj, 2025-01-29:
Googling around suggests that there isn't quite a way to make an automatic generation of the swagger document from the code. I played around a bit and at an embedded level I just don't think we are going to get the type of compiler/interpreter support to auto generate such a file. 

So then it comes down to just creating a swagger.json manually. I think that is how this is going to move forward. Again, I'm down to give it a shot. I do get the sense that these devices are so memory constrained that hosting random json/yaml files is going to push us past the limits. This is way out of my wheelhouse. 

Anyway, I still think I'm down to try to manually create the swagger doc.

netmindz, 2025-01-29:
Yeah you won't be able to use any automatic generation due to the lack of reflection in C++

There is no need to include the swagger-ui as you can just include a link to open https;//editor.swagger.io/ which will help save space 

If it's being build manually then perhaps we host the spec file from WLED-Docs, if we can pass the server URL to the editor as well as the spec

13rac1, 2025-06-16:
I've just published a tested multi-file OpenAPI 3.0 Schema for WLED 0.15.0 at https://github.com/13rac1/openapi-wled 

* [Rendered docs](https://13rac1.github.io/openapi-wled/)
* [Object schemas](https://github.com/13rac1/openapi-wled/tree/main/openapi/components/schemas/)
* [Config object schemas](https://github.com/13rac1/openapi-wled/tree/main/openapi/components/schemas/config/)

13rac1, 2025-06-18:
Nice trolling attempt 👏🏼 😆 How is this relevant to the topic? Please review https://github.com/wled/WLED/blob/main/CODE_OF_CONDUCT.md before replying.

This is how I write, full sentences with proper grammar and markdown formatting for clarity. I don't need a bot for this. Feel free to review my decade+ of Github history, there are numerous examples of my writing style. Would you like to see the history of the Google Keep Note where I originally drafted the original comment? How about my Firefox history collecting all of those links to be thorough? I've got screenshots 💻 

Nitpick: The generated code was committed with `git commit --author`. I am just the committer, not the co-author on those commits. I've renamed and reused an old DevOps-automation machine account account for this purpose.

_This text is pure artisanal human keyboarding, no AI was used!!one_ 🙃