wled/WLED #3939: Error html_ui.h when building  WLED-0_15

JonnyTech, 2024-04-28:
### What happened?

Tring to build WLED-0_15 terminates with an error
`wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory`

### To Reproduce Bug

Open source folder in VS Code
Open `platformio.ini`
Modify for required env:
```
# CI/release binaries
; default_envs = nodemcuv2, esp8266_2m, esp01_1m_full, nodemcuv2_160, esp8266_2m_160, esp01_1m_full_160, esp32dev, esp32_eth, esp32dev_audioreactive, lolin_s2_mini, esp32c3dev, esp32s3dev_16MB_opi, esp32s3dev_8MB_opi, esp32s3_4M_qspi, esp32_wrover
default_envs = esp32_eth
```
Click build
Wait for error

### Expected Behavior

Compilation should complete correctly as per instructions at https://kno.wled.ge/advanced/compiling-wled/

### Install Method

Self-Compiled

### What version of WLED?

v0.15.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
*  Executing task: platformio run --environment esp32_eth 

Processing esp32_eth (board: esp32-poe; platform: espressif32@3.5.0; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
npm run build

> wled@0.15.0-b3 build
> node tools/cdata.js

node:internal/modules/cjs/loader:1146
  throw err;
  ^

Error: Cannot find module 'inliner'
Require stack:
- /tmp/WLED-0_15/tools/cdata.js
    at Module._resolveFilename (node:internal/modules/cjs/loader:1143:15)
    at Module._load (node:internal/modules/cjs/loader:984:27)
    at Module.require (node:internal/modules/cjs/loader:1231:19)
    at require (node:internal/modules/helpers:179:18)
    at Object.<anonymous> (/tmp/WLED-0_15/tools/cdata.js:20:17)
    at Module._compile (node:internal/modules/cjs/loader:1369:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1427:10)
    at Module.load (node:internal/modules/cjs/loader:1206:32)
    at Module._load (node:internal/modules/cjs/loader:1022:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:135:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/tmp/WLED-0_15/tools/cdata.js' ]
}

Node.js v20.12.2
[... 122 lines cut ...]
*****************************************************************

compilation terminated.
*** [.pio/build/esp32_eth/src/wled_server.cpp.o] Error 1
Compiling .pio/build/esp32_eth/lib278/AsyncTCP@src-39de97abf7348c44d4dda815b8aab0ae/AsyncTCP.cpp.o
============================================================== [FAILED] Took 6.57 seconds ==============================================================

Environment    Status    Duration
-------------  --------  ------------
esp32_eth      FAILED    00:00:06.573
========================================================= 1 failed, 0 succeeded in 00:00:06.573 =========================================================

 *  The terminal process "platformio 'run', '--environment', 'esp32_eth'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```

### Anything else?

This builds correctly:
Build 2404040
WLED 0.14.3 release

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-04-28:
@JonnyTech There was a change in 15 for ui build which is automated now 
[image]

You can look at the source here  https://github.com/Aircoookie/WLED/tree/0_15 with the changes , if you compile that it will work , I do not compile on VSC but only on github

JonnyTech, 2024-04-28:
Thanks for your reply. Sorry, I am not a developer and was just following instructions the compiling guide.

So the section is there in `platformio.ini`:

```
[scripts_defaults]
extra_scripts =
  pre:pio-scripts/set_version.py
  post:pio-scripts/output_bins.py
  post:pio-scripts/strip-floats.py
  pre:pio-scripts/user_config_copy.py
  pre:pio-scripts/build_ui.py
```

Do I need to change it for compilation to succeed?

Should I be using something else to compile locally?

JonnyTech, 2024-04-28:
Looking through the rest of the documentation, I found this:

https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#changing-web-ui

```
Changing Web UI

In order to conserve space, the Web UI interface is represented as a series of wled00/html_ui.h, wled00/html_settings.h and wled00/html_other.h files which contain C/C++ strings with specific parts of the Web UI.

These files are automatically created from source files available in wled00/data folder. To generate files, install [NodeJS 11.0+](https://nodejs.org/en/download/) globally. After that, recreate html_*.h files by running in the repo directory:

> npm install
> npm run build
```

After running those two commands, it now compiles correctly.

When I have a moment, I shall create a PR for the documentation.

closed 2024-04-28 by JonnyTech

blazoncek, 2024-04-28:
Wiki is no longer used.

JonnyTech, 2024-04-28:
> Wiki is no longer used.

Thanks @blazoncek.

Yes, indeed, as stated in my first post, the guide was not on the wiki:

https://kno.wled.ge/advanced/compiling-wled/

But the tip to execute the `npm` commands was found on the wiki.

Aircoookie, 2024-04-28:
`npm install` is the last item in the "Preparation" section on https://kno.wled.ge/advanced/compiling-wled/

Perhaps it is a bit easy to miss, and I would love for npm to run install automatically on first build in a fresh environment, so that step can be omitted. Maybe I can figure something out.

JonnyTech, 2024-04-28:
Thanks @Aircoookie - my bad, I completely missed it.

While you are here, the reason that I am trying to recompile WLED is so that I can adapt the serial API to accept cfg JSON so that I can change the IP address. Is this possible at all?