wled/WLED #3941: WLED control via USB/OpenRGB

MARK2580, 2024-04-29:
### What happened?

The documentation says that the latest WLED builds support control via Serial Port. I'm trying to connect to WLED via a regular USB cable and feed data to the COM port using the OpenRGB program.
When the program starts sending data to the COM port, the WLED interface completely stops working, and sometimes the board simply freezes, only turning off the power helps.

### To Reproduce Bug

board - WEMOS D1 mini
WLED - 0.15.0-b1
[image]

### Expected Behavior

I expected that WLED would work similarly to, for example, an arduino board with code for working with Adalight.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b1 (build 2403190)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-04-29:
WLED supports Adalight protocol and Improv over serial.

MARK2580, 2024-04-29:
> WLED supports Adalight protocol and Improv over serial.
> 
> EDIT: Adalight should work normally (you can verify if you can see "Ada" displayed over serial). If you have debug build, Adalight might be unavailable.

All I have is an ESP and the OpenRGB program; unfortunately, I am far from programming. 🙄

MARK2580, 2024-09-01:
As far as I understand, no one plans to fix this problem??? I even tried it on ESP32 with WLED 0.15.0-b4 firmware and everything I described above is repeated exactly. The board simply hangs until it stops receiving a signal via the COM port.

SellA, 2024-12-23:
Got exactly the same behaviour on an ESP32 Wroom and WLED 15.0 . I tried also to open a console from putty and sending 'v' character. I see light on the serial LED on the board but no output to the serial window.