wled/WLED #3947: Panel color intensity inconsistent across strips

bandi13, 2024-05-01:
### What happened?

Switching between `main` and `0_15` builds, I notice that the color brightness across panels is wildly varying in `0_15`. You can see in [main branch](https://youtu.be/liHmAcVCMWM) it works correctly, whereas in [0_15](https://youtu.be/K-AWEstq0HA) it does not. Particularly, the middle panels are affected in the videos, but it's consistently random throughout.

### To Reproduce Bug

Build `main` using commit 197f47befe568eef486d006aaec50b06c11c5fc4. Confirm that it works using [parseVid.py](https://github.com/bandi13/LEDMatrix/blob/main/parseVid.py).

Then build `0_15` using commit 8110259d1cd79c61c1ea633ac89c83573445d939. Run `parseVid.py` again and notice that the segments are different intensities.

The configuration on the ESP32 was not modified, only new firmware was uploaded. The issue is exactly reproducible.

### Expected Behavior

Expectation is that the panels have a uniform intensity.

### Install Method

Self-Compiled

### What version of WLED?

'main' vs '0_15'

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I used the same configuration as [here](https://github.com/Aircoookie/WLED/issues/3922#issuecomment-2078518937).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-05-01:
I do not have 48x48 setup so not sure how to replicate that if that is part of the issue .
Power might also  be a factor in such large setup and not sure you have tried the same with 
ABL off as in other cases something like that was solved when ABL was turned off  .

Still testing on other smaller setups I do not see any color intensity issue.
I have tested using jinx and might be it is better that you use that or xlights if you are on mac
as i got lots of issues with your script ( you have not included the files you are using to test on your repo ) 
which might be the cause of the issue but in all cases testing with a well known app might be 
better to as that could be replicated by others to eliminate the guess work .

softhack007, 2024-05-01:
I agree with @dosipod it could be caused by ABL (auto brightness limiter). You can disable this function in LED settings.

One difference in 0_15 is the new capability to use ABL "per led pin" while in 0.14.x the feature was always running globally (= reducing brightness in the same way on all outputs). So if your "middle panel" runs on it's own LED pin, then what you see is most likely "expected behaviour".

blazoncek, 2024-05-01:
As mentioned disable ABL.

bandi13, 2024-05-02:
In my setup each panel has its own pin for a total of 9 pins used. Disabling the ABL does make each panel's brightness identical in 0_15 but means to me a feature that was working before no longer works.

I had the "per output limiter" checkbox disabled. Wouldn't that mean the ABL is treated on a global scale?

As for the testing script, it's a pretty simple Python program. If you can't get OpenCV or its dependencies on a Mac, then consider running the script in a container.

blazoncek, 2024-05-02:
ABL, even though you uncheck "per output" option, still works per output. It just proportionally distributes available current to each output depending on its length.

So, if the brightness on each individual output would cause overcurrent,that output's brightness is reduced but not the others. That is as it is designed.

If you would want it the other way, that particular output (with overcurrent) could fry your PSU for that output.

Unfortunately each output behaves individually and since per output ABL is only possible at bus level, there is no other way to accomodate your desire, unless some convoluted approach is made.
If you can figure out a simple/trivial way you are welcome to open a PR.

softhack007, 2024-05-02:
@blazoncek idea: add a mode where all outputs compute their own ABL down-scaling factor, then take the lowest (=strongest) from all busses and apply to all outputs. This would ensure uniform dimming while not risking burning cables. Downside: strips might go to lower brightness than technically needed.

blazoncek, 2024-05-02:
> idea: add a mode where all outputs compute their own ABL down-scaling factor, then take the lowest (=strongest) from all busses and apply to all outputs.

Unfortunately ABL is calculated during single output's `show()`. Once it is done, there's nothing more you can do. LEDs are already updating. Doing it in multiple passes would only slow things down.

For the OP case I would recommend either: a) turn ABL off and use adequate PSU or b) use per port ABL and set appropriate current (overprovisioned).
With b) you will need to make sure never to use white at full brightness on all pixels.

There is a possibility to use summed ABL current on all outputs but that just silently calls for disaster so I prefer the way it is even though it means some panels will be dimmer than others. If that is an issue there is always option a).

closed 2024-09-08 by github-actions[bot]