wled/WLED #3948: Struggling to install on ESP32-S2 M1 DevKit

Babyyoda777, 2024-05-01:
### What happened?

Hi i am trying to install WLED for a project on ESP32-S2 dev kit board. I have tried everything, the various web installers just don't work. The official one gives header error whilst the unofficial one says no core dump partition found. I tried flashing myself using es32-s2 binary and esptool along side a bootloader I compiled and partition map however that also didnt work. When compiling with platformio, it doesnt have the same ESP32-S2 board Im using and when i try using saloa esp32s2 board platformio just refuses to build with error: Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' requirements for your system 'darwin_x86_64'

Would really appreciate some help as ive become very lost. Thanks!

### To Reproduce Bug

Attempted to install using all suggested methods in docs, doesnt work.

### Expected Behavior

Installs and boots normally

### Install Method

Binary from WLED.me

### What version of WLED?

Latest official release (not beta)

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-05-01:
Hi,

Please consider that the new MCUs (-S3/-S2/-C3) are still experimental - so w boards work, some don't work.

The best way is actually to compile _and upload_ with platformIO.

You can use this build environment:
https://github.com/Aircoookie/WLED/blob/197f47befe568eef486d006aaec50b06c11c5fc4/platformio.ini#L617

* the `board = ` doesn't need to match exactly with your board, lolin sw mini will work with most S2 boards
* "no core dump partition found" is just a warning, you can ignore that one.
* if you struggle to find the "build & upload" in VSCode+platformIO IDE, there are some how-to screenshots here: https://github.com/Aircoookie/WLED-Docs/issues/161

softhack007, 2024-05-01:
PS: there is ongoing work to make the official web installer work with more boards. For flashing manually with esptool, you can first a list of items needed here https://github.com/Aircoookie/WLED-WebInstaller/issues/22#issuecomment-2080430946

Babyyoda777, 2024-05-01:
Thanks so much, I'll give it all another go!

blazoncek, 2024-05-01:
I'm closing as installing WLED firmware is a separate topic. WLED itself works correctly on S2, S3 and C3, albeit not on *every* model there is.

closed 2024-05-01 by blazoncek

Babyyoda777, 2024-05-02:
Thanks I managed to sort it out! I created a repo with compiled binaries and instructions for installation using web gui.